danielnilsson9/bbs-fw #2: Support for a chainforce torque sensor

freisei, 2021-06-22:
Hi!

i want to build/develop a simple mountable sensor to add a torque-sensor for the BBSHD motor. My Idea is to basicaly measure the chain-tension, substract the power given from the motor. The result should be the power given from pedals.

Your firmware/projekt is nearest one i to production state i found, so do you think you can (and want) add software functionality for this idea? 
When yes: Which type of signal do you need from the sensor? 

At the moment i am to far away from c, that i could do this by my own in a quality that someone other can work with. But if the basic code is running i can do programming or debugging.

Greets from Bavaria

droidi, 2021-06-23:
Hey
So sorry. I am just looking for an open firmware to get 2 different settings.
Just programming / compiling arduino so far.
Wish to get a ready compiled .bin or .hex fitting to the 1.5  version  of the bbshd.

Sorry for my bad english and 
good luck for your project.

danielnilsson9, 2021-06-24:
I'm not actively working on this firmware at the moment as I have been unable to figure out why it seems impossible to get a stable throttle reading even though it works with the original firmware. 
I will therefore not provide binaries since it's not really safe or usable at the moment. But check wiki if you want to compile it yourself.

As for interfacing with torque sensors, you will need some way to get the signal into the BBSHD controller. 
There are two additional UART:s brought out to pins holes on the PCB which could be used but would require depotting and soldering wires to the controller. An alternative way could be to connect a UART relay proxy on the display connector where you could inject custom messages from the sensor, which would probably be the easiest and cleanest way.

When you have you sensor reading, check app.c app_process() where motor current is set after calculating everything, given in percent, 0-100: motor_set_target_current(value).

freisei, 2021-06-28:
>  to get a stable throttle reading 

Which throttle do you mean here? The throttle sent from thumb-trottle-sensors, cadence-sensor or do you mean the power/current-consumption from motor?

> As for interfacing with torque sensors, you will need some way to get the signal into the BBSHD controller.
> There are two additional UART:s brought out to pins holes on the PCB which could be used [...]

At the moment i use the original-firmware and i am a "man in the middle" between display and motor. This is really easy and i can imagine that this is a good solution to develop a end-user ready product. Of course a implementation in display-firmware would be the most pretty - lets see how your firmware is going on...

> 
> When you have you sensor reading, check app.c app_process() where motor current is set after calculating everything, given in percent, 0-100: motor_set_target_current(value).

Many thanks for your informations!

danielnilsson9, 2021-08-05:
> Which throttle do you mean here? The throttle sent from thumb-trottle-sensors, cadence-sensor or do you mean the power/current-consumption from motor?

Reading from physical thumb throttle. Motor power set from e.g. PAS is stable.

> At the moment i use the original-firmware and i am a "man in the middle" between display and motor. This is really easy and i can imagine that this is a good solution to develop a end-user ready [...]

Sound like a good idea, have you continued work on this? Which hardware sensor did you intend to use?

danielnilsson9, 2022-07-27:
Closing this as it is more of a discussion. The problems described above has now been fixed.
If there is more to be said on the topic of the issue, open a thread in the discussion section.

closed 2022-07-27 by danielnilsson9