danielnilsson9/bbs-fw #3: Com port not found

4FF, 2021-06-27:
Does not find CH340 USB interface and add to the COM ports selection.

4FF, 2021-06-29:
```
public static List<ComPort> GetComPorts()
		{
			var result = new List<ComPort>();

			using (var searcher = new ManagementObjectSearcher("SELECT * FROM WIN32_SerialPort"))
			{
				var portNames = SerialPort.GetPortNames();
				var ports = searcher.Get().Cast<ManagementBaseObject>().ToList();

				foreach (var name in portNames)
				{
					var port = ports.FirstOrDefault(p => String.Equals(p["DeviceID"].ToString(), name));
					if (port != null)
					{
						result.Add(new ComPort(name, port["Caption"].ToString()));
					}
					else
					{
						result.Add(new ComPort(name, "NAME GATHERING FAILED"));//fix
					}
				}
			}

			return result;
		}
```
Fixes the connection issue, but then if connection succeeds it timeouts on read flash.
But for some reason if stepping trough serial read function with debugger it works fine.
Perhaps some delay needed somewhere.
Problematic interface  "USB SERIAL CH340"

fpdragon, 2021-08-03:
Same problem here. I was able the flash the controller but the same USB to UART adapter is not detected by the config tool.

I then tried to tune the default values of the firmware to get my configuration.

in crgstore.c I changed the following:
g_config.max_current_amps = 25; // I don't want to overload the controller
g_config.low_cut_off_V = 45; // 45V for a 15s battery (60V max)
g_config.wheel_size_inch_x10 = 290; // for a 29 inch wheel size
g_config.max_speed_kph = 35; // 35kph is just enough for me

What happened is that the display (500C) shows an error 04h.
The displayed speed seems to be correct. Brake and shift sensors seems to be not working.
The walk mode seems to work.
When I start paddeling the motor kicks in very hard. Even on the 1st assistant level is set it is always the same and the motor starts pretty powerfull for the first seconds. Then after some seconds and a speed of ~22kph the motor stops running allthough it should run up to 35kph. When I continue driving by muscle I see sometimes that the displays shows some litte wattage but I don't feel any assistance at all.

danielnilsson9, 2021-08-03:
Makes sense, an unfortunate oversight. Please send me a pull request with your fix

fpdragon, 2021-08-04:
I also made some modifications in the COM listing and now it seems that I am able to connect to the controller.

```
		public static List<ComPort> GetComPorts()
		{
			var result = new List<ComPort>();

			//using (var searcher = new ManagementObjectSearcher("SELECT * FROM WIN32_SerialPort"))
			using (var searcher = new ManagementObjectSearcher("SELECT * FROM Win32_PnPEntity WHERE Name LIKE '%(COM[0-9]%'"))
			{
				var portNames = SerialPort.GetPortNames();
				var ports = searcher.Get().Cast<ManagementBaseObject>().ToList();

				foreach (var name in portNames)
				{
					string sName = (string)ports[0].Properties["Name"].Value;
					string sValue = (string)ports[0].Properties["DeviceID"].Value;
					//var port = ports.FirstOrDefault(p => String.Equals(p["DeviceID"].ToString(), name));
					var port = ports.FirstOrDefault(p => p["Name"].ToString().ToLower().Contains(name.ToLower()));
					if (port != null)
					{
						result.Add(new ComPort(name, port["Caption"].ToString()));
					}
				}
			}

			return result;
		}
```

But as 4FF stated there is still a timeout when trying to read the flash. I agree that it may be debugged and it's maybe just a delay that needs to be added to get the buffers filled in time. Or the communication handling itself needs to be changed to be able to handle the protocol with different timings.

danielnilsson9, 2021-08-04:
Just to confirm, you also have a CH340 based usb to serial adapter?

danielnilsson9, 2021-08-04:
If you start config tool in debug from visual studio, do you get any output in the visual studio console?
Something like "Discarding ...."?

4FF, 2021-08-04:
8

4FF, 2021-08-04:
Its likely that most bafang user base have 50 cent "USB to TTL" also called ch340 or ch341 interface for connection ,its sold with cable for ~20USD. 
Might get interface separately in electronics store and use preexisting cable , otherwise shipping from china takes time.

Useful to have same device that users usually have.

Before testing the firmware was running original 52V firmware with modification to allow up to 40 amp parameter in bafang tool.
If setting 1 or more amps over 33 it goes crazy and throttle became on/off switch while controller limits power to only half.

Compared to 48V firmware 52V version prioritized pedals and was not smooth if using throttle while pedaling.

When i flashed your firmware didnot get smooth power for some reason, it was looping power on/off on higher loads and 50% battery.
Used config tool and set low voltage cutoff under 38 and some other parameters out of ranges perhaps, maybe that was the cause.
My goal was to make throttle smoother compared to original and use max amps possible.

Later i modified 48V original firmware instead to gain some amps and currently use that firmware as throttle is prioritized over pedals and in amp mode with right settings throttle is quite smooth saving drivetrain from hits.

Think this project could improve alot in future  if enough time put in.

fpdragon, 2021-08-04:
> Just to confirm, you also have a CH340 based usb to serial adapter?

Yes, I have a CH340 adapter.

4FF, 2021-08-04:
> Not sure if you had the speed sensor enabled or not in this firmware when you tested, if so that may have caused some weird behaviour that should be fixed now.
> 
Yes, was enabled.
> BUT, throttle still has issues which I cannot find a fix for, once every minute or so there are false readings making it go 100% for a short period so this firmware is still not useful with throttle.
> 
Didnot notice this problem.
With my settings it seemed to reduce power instead and at ~1-3 sec frequency if using high throttle input.

fpdragon, 2021-08-04:
I also tested this with latest modifications that we (daniel and me) have made regarding speed limiting.
I guess it's not a firmware problem and the config app has to be adapted for the different timing.
But that's just a guess...

danielnilsson9, 2021-08-04:
> > Not sure if you had the speed sensor enabled or not in this firmware when you tested, if so that may have caused some weird behaviour that should be fixed now.
> 
> Yes, was enabled.
> 
> > BUT, throttle still has issues which I cannot find a fix for, once every minute or so there are false readings making it go 100% for a short period so this firmware is still not useful with throttle.
> 
> Didnot notice this problem.
> With my settings it seemed to reduce power instead and at ~1-3 sec frequency if using high throttle input.
> Motor board version was the later one.
> 
> But i didnot test on the road with initial settings before using config tool and making changes.

Alright, sounds much like the behaviour of the broken speed limiting then, but that should be fixed now at least.

fpdragon, 2021-08-04:
I have just retested the config tool with the newest firmware fixes and reading flash still fails.

danielnilsson9, 2021-08-04:
Please send me output from from visual studio console when you try to read flash, if possible.

danielnilsson9, 2021-08-05:
Received my CH340 and was able to reproduce the error, will investigate.

danielnilsson9, 2021-08-05:
Should be fixed now, it was timing related. The detection of failed communication was to fast, this chip seems to buffer received data for a longer period.

closed 2021-08-24 by danielnilsson9