danielnilsson9/bbs-fw #4: Firmware kicks in hard and then stops at 20kph

fpdragon, 2021-08-03:
Since the config tool can't connect over my USB to UART cable, I tried to tune the default values of the firmware to get my configuration.

in crgstore.c I changed the following:
g_config.max_current_amps = 25; // I don't want to overload the controller
g_config.low_cut_off_V = 45; // 45V for a 15s battery (60V max)
g_config.wheel_size_inch_x10 = 290; // for a 29 inch wheel size
g_config.max_speed_kph = 35; // 35kph is just enough for me

What happened is that the display (500C) shows an error 04h.
The displayed speed seems to be correct. Brake and shift sensors seems to be not working.
The walk mode seems to work.
When I start paddeling the motor kicks in very hard. Even on the 1st assistant level is set it is always the same and the motor starts pretty powerfull for the first seconds. Then after some seconds and a speed of ~22kph the motor stops running allthough it should run up to 35kph. When I continue driving by muscle I see sometimes that the displays shows some litte wattage but I don't feel any assistance at all.

Hope someone can help.

danielnilsson9, 2021-08-03:
Yes, speed limiting does not work and behaves very weird, I have not investigated further. 

Please disable speed sensor:
g_config.use_speed_sensor = 0

04h is throttle error, maybe you don't have one connected at all? Looks like there is a bug where this error will be reported if no throttle is connected.

Anyway, as stated this firmware is not really in a stable working condition so I don't recommend anyone using it at the moment.

danielnilsson9, 2021-08-03:
Try tweaking the assist levels and see if that helps default might be to high:

uint8_t current_limits[9] = { 25, 34, 43, 51, 60, 68, 74, 82, 90 };

values are in % [0-100]

danielnilsson9, 2021-08-03:
You can remove the throttle error code by editing thottle.c 

replace:
```
bool throttle_ok()
{
	return throttle_low_ok && throttle_hard_ok;
}
```

with:
```
bool throttle_ok()
{
	return true;
}
```

fpdragon, 2021-08-03:
Thank you very much! :-D
But how will this work when the speedsensor is not working? I guess this would make it hard to ride with a constant speed

fpdragon, 2021-08-03:
And of course you were right. I have no throttle sensor connected.

danielnilsson9, 2021-08-03:
Yes, I mean this firmware is not in any way ready for everyday use, only for testing and development. So if you need that I would recommend using the original firmware.

fpdragon, 2021-08-03:
Fair enough. No release SW.

As you said:
throttle_ok = true removes the error.
disabling the speed sensor seems to normalize the whole behavior at all.
brake sensors seem to work now.
Not sure about the shift sensor since it blinks red and green xD
I reduced the first current limit from 25% to 5% (from 25A) and still rocket mode xD.

I still really like your work and it's awesome that there is this open source project. 
My main problem is that I want to use the BBSHD with a 15s pack (3x 5s Makita batteries). With the Bafang firmware I get an error 07h when I all batteries are full charged and also the low voltage cutoff is way to low.
I do noticed that this could be resolved by making a HW modification that reduces the voltage on the display power line (brown cable). With a resistor or diodes.
However I found it much more interesting to upload a firmware that supports the voltage out of the box, what your code does. That's really nice. I also could set a undervoltage protection to 45V which is not possible with the Bafang firmware but seems to be possible with your SW.

However I guess the show stopper is the lack of the speed sensor support. I guess I have to do the HW mod to get my bike running for daily use.

Maybe I look into your code a little bit deeper. Maybe we could solve the problem with the speed sensor? What do you think? Is it a low hanging fruit or is it still a hard way to go until the speedsensor will work?

danielnilsson9, 2021-08-03:
Ahh I see, glad I got that tested then (15S), only assumed it would work but you have now confirmed that then, thanks!

My guess is that the speed sensor thing should be something pretty simple, I think the actual speed reading is correct on display or do you think its incorrect?

I have been fully focused on the throttle issues so I haven't looked on the speed limiting issue at all, only noticed it.

Speed limiting is applied in app.c:

```
void apply_speed_limit(uint8_t* target_current)
{
	if (g_config.use_speed_sensor && assist_max_wheel_speed_rpm_x10 > 0)
	{
		int16_t current_speed_x10 = speed_sensor_get_rpm_x10();

		int16_t high_limit = assist_max_wheel_speed_rpm_x10 + 20;
		int16_t low_limit = assist_max_wheel_speed_rpm_x10 - 20;
	
		if (current_speed_x10 < low_limit)
		{
			// no limiting
			if (speed_limiting)
			{
				speed_limiting = false;
				eventlog_write_data(EVT_DATA_SPEED_LIMITING, 0);
			}
		}		
		else
		{
			if (current_speed_x10 > high_limit)
			{
				if (*target_current > 2)
				{
					*target_current = 2;
				}
			}
			else
			{
				// linear ramp down when approaching max speed.
				uint8_t tmp = (uint8_t)MAP(current_speed_x10, low_limit, high_limit, *target_current, 2);
				if (*target_current > tmp)
				{
					*target_current = tmp;
				}
			}

			if (!speed_limiting)
			{
				speed_limiting = true;
				eventlog_write_data(EVT_DATA_SPEED_LIMITING, 1);
			}
		}
	}
}
```

If reading on display seems to be correct something must be wrong in this function.

fpdragon, 2021-08-03:
I can confirm that 15s is working well without any error 07h. And that's without any HW modification. Awesome!

The speed reading on the display seems to be in a correct range at least but I have not compared it to a second speed sensor. The only thing I have noticed that if you stop the rotating wheel abruptly then the last speed reading stays on the screen for a long time. But this could be normal due to a large timeout for slower speed measurements. However, if the hall sensor gets no magnet reading for seconds than it can't be that the speed is still above 20kph. So at least the displayed value could be reduced as a best guess.

I played with the current limits and also set very low levels like 1 - 5%. I can confirm that these values have an effect on the power that gets on the wheel. A kind of power ramp up would definitely make sense here. Currently it seems that I would have to start riding with a low level and then switch to a higher level while riding. When starting on a high power assistance then the bike nearly jumps up. The kick in is hard but when driving it would make sense to rise the power a little bit. But I guess a power ramp up is a thing that can be easily extended when the SW gets stable. 

The more critical thing is the speed sensor support. At my testrides the speed gets totally out of control and you permanently stop paddeling or switching between the assistance levels.

To the throttle, I do have a throttle but I have not installed it. In my country throttles are not alowed and I still want to paddle on my bike a little bit. But for testing I could connect the throttle if you need something.

fpdragon, 2021-08-03:
have you tested this function?
`uint16_t convert_wheel_speed_kph_to_rpm(uint8_t speed_kph)
{
	float radius_mm = g_config.wheel_size_inch_x10 * 2.54f;
	return (uint16_t)(25000.f / (3 * 3.14159f * radius_mm) * speed_kph);
}
`

fpdragon, 2021-08-03:
For example:
`		assist_max_wheel_speed_rpm_x10 = convert_wheel_speed_kph_to_rpm(6) * 10;
`
I guess this is the walk mode that is set to 6kph.
However when I activate walk mode it seems that it ramps up to 10kph. But maybe it's again because the motor still has to much power and jumps above the 6kph goal. But as far as I see it seems a little bit strange.

fpdragon, 2021-08-03:
And the postfix _10x also confuses me a little bit.
Sometimes these variables are 10 times higher and sometimes 10 times lower?
For example:
```
		assist_max_wheel_speed_rpm_x10 = ((uint32_t)global_max_speed_rpm * assist_level_data.max_speed_percent / 10);
		assist_max_wheel_speed_rpm_x10 = convert_wheel_speed_kph_to_rpm(6) * 10;

```
one time rpm/10 and one time rpm*10 and it is saved to the same variable.

However in this case I guess it's correct since the max_speed_percent is a 100x factor. So it is:
`		assist_max_wheel_speed_rpm_x10 = ((uint32_t)global_max_speed_rpm * assist_level_data.max_speed_percent / 100 * 10);
`

danielnilsson9, 2021-08-03:
> The speed reading on the display seems to be in a correct range at least but I have not compared it to a second speed sensor. The only thing I have noticed that if you stop the rotating wheel [...]

I remember that too but I'm pretty sure it is the display firmware that causes that. I'm pretty sure I observed the same behavior with the original firmware. I believe the value staying on for a long time is some averaging implemented in the in the display firmware.

The speed calculation is done in the display, only wheel rpm is sent from controller firmware to display. So you must configure the correct wheel size in the display for correct speed presentation.

> I played with the current limits and also set very low levels like 1 - 5%. I can confirm that these values have an effect on the power that gets on the wheel. A kind of power ramp up would [...]

Yes, that is something that needs to be implemented, should be pretty straight forward.

> To the throttle, I do have a throttle but I have not installed it. In my country throttles are not allowed and I still want to paddle on my bike a little bit. But for testing I could connect the throttle if you need something.

Would be interesting to confirm that others observe the same behavior as I do (throttle going crazy for split second when holding steady, happens every now and then). If you test, maybe configure lower max amps since throttle will go instantaneous to 100% for short intervals.

danielnilsson9, 2021-08-03:
> have you tested this function?
> 
> ```
> uint16_t convert_wheel_speed_kph_to_rpm(uint8_t speed_kph)
> {
> 	float radius_mm = g_config.wheel_size_inch_x10 * 2.54f;
> 	return (uint16_t)(25000.f / (3 * 3.14159f * radius_mm) * speed_kph);
> }
> ```
> 
> I'm not sure what comes out when mixing the types in the calculation without casting.

I'm not sure I have, it could be wrong. But integers are automatically upcasted to float when multiplied by a float so that should be ok.

> And the postfix _10x also confuses me a little bit.
> Sometimes these variables are 10 times higher and sometimes 10 times lower?
> For example:
>
> ```
> 		assist_max_wheel_speed_rpm_x10 = [...]

That seams to be correct yes. I agree, it is all a bit confusing. There is a lot of fixed point maths here for execution efficiency and for the extremely limited stack space on this microcontroller.

danielnilsson9, 2021-08-03:
I do release that there is something wrong in 
void apply_speed_limit(uint8_t* target_current)

+20 and -20 in high_limit and low_limit where intended to represent 2.0km/h. I.e. start power ramp down 2km/h before reaching max speed. But these variables represent rpm so that is completely wrong.

fpdragon, 2021-08-03:
> I do release that there is something wrong in
> void apply_speed_limit(uint8_t* target_current)
> 
> +20 and -20 in high_limit and low_limit where intended to represent 2.0km/h. I.e. start power ramp down 2km/h before reaching max speed. But these variables represent rpm so that is completely wrong.

Ahhh I already was wondering why +-2rpm threshold is set. however this does not explain the effects I have seen while riding with speed sensor on.

danielnilsson9, 2021-08-03:
> have you tested this function?
>
> ```
> uint16_t convert_wheel_speed_kph_to_rpm(uint8_t speed_kph)
> {
> 	float radius_mm = g_config.wheel_size_inch_x10 * 2.54f;
> 	return (uint16_t)(25000.f / (3 * 3.14159f * [...]

This function is wrong.

It should be :
radius_mm = g_config.wheel_size_inch_x10 * 2,54 / 2.f;

wheel_size_inch is diameter should divide by two for radius.

danielnilsson9, 2021-08-03:
If you want to test right now, replace 20 with 140 and fix the convert_wheel_speed_kph_to_rpm function.

fpdragon, 2021-08-03:
`				uint8_t tmp = (uint8_t)MAP(current_speed_x10, low_limit, high_limit, *target_current, 2);
`
This also confuses me...

As far as I see, there is this MAP macro that uses two ramps to find an corresponding point from the input ramp to the output ramp.
So we have 5 values.
current_speed_x10 -> input point value
low_limit, high_limit -> start/end values for the input compare ramp
*target_current, 2 -> start/end ramp values for the output value calculation

inputs are all rpm x10
outputs are all current

the input ramp starts at low_limit that corresponds the rpm goal - 2 rpm (later -2kph)
the output ramp should correspond the input ramp.
output ramp start with the current amp that is currently pulled by the motors? why should this correspond to rpm goal -2 rpm?
And the output ramp end is at 2 amp. Why that? So when the max rpm is reached then still a little 2 amp shall be pulled?

What I would expect is that:
at the lower rpm limit I would expect the maximum possible amp that is configured in this mode.
in the middle of the ramp there is a point where the target_current should stay as is since the target rpm is reached.
and coming to the high limit rpm the current should be reduced. is the high limit is exceeded then I would expect to completely disable the motor. (eg riding downhill)

But maybe I missunderstood something here?

fpdragon, 2021-08-03:
> > have you tested this function?
> > ```
> > uint16_t convert_wheel_speed_kph_to_rpm(uint8_t speed_kph)
> > {
> > 	float radius_mm = g_config.wheel_size_inch_x10 * 2.54f;
> > 	return (uint16_t)(25000.f / (3 * 3.14159f * radius_mm) * speed_kph);
> > }
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > I'm not sure what comes out when mixing the types in the calculation without casting.
> 
> This function is wrong.
> 
> It should be :
> radius_mm = g_config.wheel_size_inch_x10 * 2,54f / 2.f;
> =>
> radius_mm = g_config.wheel_size_inch_x10 * 1.27f;
> 
> wheel_size_inch is diameter should divide by two for radius.
> 
> I suspect both these errors may cause the weird behavior.

Cool. I promise I will give it a try tomorrow. It's pretty late here in europe.

Do you have any possibility to debug the code? Any diagnostic output or so?

danielnilsson9, 2021-08-03:
> ` uint8_t tmp = (uint8_t)MAP(current_speed_x10, low_limit, high_limit, *target_current, 2); `
> This also confuses me...
>
> As far as I see, there is this MAP macro that uses two ramps to find an [...]

You are pretty much correct, MAP maps a value in an input range to a desired output range.

So the logic is, as input approaches high_limit output should approach 2. Do not remember why 2 was set there, could probably be 0 instead but I guess I didn't want to completely disable the motor as that might feel weird when you are right on high_level and it may turn on/off rapidly. But that may need to be tweaked.

target_current is not something that is measured, it is what we want to send to the motor and it is computed by several factors.
See app_process()

Basically pas, throttle and cruise will contribute to this value. and apply_speed_limit() and apply_thermal_limit() may constrain it. You can see the order of calculations in app_process().

So if you have no throttle and are not using cruise, only PAS and no thermal limiting takes place then target_current will be equal  to maximum set PAS level current.

danielnilsson9, 2021-08-03:
> Cool. I promise I will give it a try tomorrow. It's pretty late here in europe.
>
> Do you have any possibility to debug the code? Any diagnostic output or so?

Yees same here. Let me know how it goes and I will try to fix this tomorrow.

Unfortunately I don't, only the debug log in the config tool which you are not able to use at the moment.
If a need to read a value while riding I usually send it to the display in the "RANGE" field.

See process_bafang_display_read_range() in extcom.c. Currently controller temperature is sent in this field.

fpdragon, 2021-08-04:
I made a testride with the latest modifications we talked about:
radius vs diameter fix
140 for about 2kph +/-

Beside this I set the following settings:
```
	g_config.max_current_amps = 20;
	g_config.use_speed_sensor = 1;
	g_config.max_speed_kph = 35;
	uint8_t current_limits_arr[9] = { 10, 20, 30, 40, 50, 60, 80, 90, 100 };
```

Now the controller does not go crazy and acts more normal.
But still... it seems that there is no speed limit and I was able to accelerate up to 40kph. After this I always stopped since I don't want to excite my neighbourhood. ;-)

Next I will try the following:
reduce max speed:
`	g_config.max_speed_kph = 20;
`
reduce the ramp down amp to 0.
```
			if (current_speed_x10 > high_limit)
			{
				if (*target_current > 0)
				{
					*target_current = 0;
				}
			}
			else
			{
				// linear ramp down when approaching max speed.
				uint8_t tmp = (uint8_t)MAP(current_speed_x10, low_limit, high_limit, *target_current, 0);
				if (*target_current > tmp)
				{
					*target_current = tmp;
				}
			}
```

So let's do it. :-)

fpdragon, 2021-08-04:
Even with the latest changes (20kph and 0 amp target on upper ramp) no speed limitation at all. I was at 38kph.

fpdragon, 2021-08-04:
I made an other investigation:
I found out that in walk mode the motor cuts off at exactly 8kph!

So without load it ramps up to 10kph. Then it slowly runs out without motor until 7.9kph and the motor kicks in.

That's unexpected since the walk mode is programmed for 6kph as you see in
`		assist_max_wheel_speed_rpm_x10 = convert_wheel_speed_kph_to_rpm(6) * 10;
`

So either the function `convert_wheel_speed_kph_to_rpm` has still a bug or the value of `assist_max_wheel_speed_rpm_x10` is not processed correctly?

However assist_max_wheel_speed_rpm_x10  seems to be only used in the `apply_speed_limit` function what makes me a little bit wonder how this line makes sense:
`		assist_max_wheel_speed_rpm_x10 = convert_wheel_speed_kph_to_rpm(6) * 10;
`

This is why I am now thinking that the walk mode is executed in `app_process` with the line
`			motor_set_target_speed(40);
`
Which would make even more sense that the walk mode should run at 4kph. However the real limit seems to be still 8kph.

To wrap it up... I am still confused...
It seems that there are two mechanisms in the source code that are limiting the speed.
One uses the global var `target_speed` that is set in kph.
This variable seems to be used for communicating with the display
`send_request(OPCODE_TARGET_SPEED, target_speed);`

The second mechanism is limiting the rpm and uses the variable `assist_max_wheel_speed_rpm_x10`.
This variable should be used in the function `apply_speed_limit` but currently it seems to have no function.

I throw in the unqualified question: Why are there two mechanisms for speed limiting?

So what I see is that the walk mode seems to be double (if 4kph) or at least 1/4th faster (for 6kph) as it should be. If the value 4kph is the value that should have a function then the limit is double so when I set the nomal assist limit to 20kph then it may be set in reallity to 40kph? I never reached this speed so maybe there is a speed limit and I just was too slow to reach it?

I will test this theory later by reducing the speed limit to 10kph that should limit it to 20kph if I am right...

danielnilsson9, 2021-08-04:
Speed limiting without any load will not work at all at the moment since we only control current and not motor rpm, so that is expected.

Sounds like it almost work as expected for push walk? If speed limit is 6kph motor will start ramping down current at 4kph and completely stop at 8kph. 

So the motor logic is run from another microcontroller which we are not touching with this firmware.
So protocol here:
https://github.com/danielnilsson9/bbshd-fw/wiki/NEC-Motor-Control-Protocol

It has two ways of control, speed (cadence) and current.

Motor speed is a value in the range [0-255] while current is in percent [0-100].

This firmware does not use motor speed (cadence) control at the moment and it is set to 255 (maximum) for all assist levels except push walk where it is set to something low (40).

So push walk is:
* Set a low current (10%)
* Set allow maximum cadence (40)
* Apply current limiting in apply_speed_limit to keep 6kph.

So why is maximum cadence set low here? Well when in push walk the load on the motor will be low and current limiting may not be enough as a low current can still make the motor spin to fast.

I have not tested this so may need some tweaking, I may also need to implement motor speed (cadence) limiting in apply_speed_limit if current limiting is not enough.

assist_max_wheel_speed_rpm_x10 is set in reload_assist_params() which is called each time you change assist level or activate push walk.

So yes, it is a bit confusing as there is three parameters to speed limiting:

* Road speed
* Motor speed (cadence)
* Motor current 

Right now we limit motor current based on road speed except for push walk where motor speed is also set to a fixed low value.

Does this make any sense?

fpdragon, 2021-08-04:
I am totally sure everything you wrote makes sense!

I have made progress. Now speed limiting seems to work. :-D
I am not sure what made it happen but I am happy that it does what it should. 
Maybe I should have power cycled the controller and I had some old version still in the ram or something xD.

So it seems that the latest modifications have worked and now the pas mode runs on exactly the configured speed. nice!

Still the walk mode seems to be confusing for me. Maybe it's because 6kph is set and the motor starts helping +2kph above and so I am on 8kph without load? That could make sense.

danielnilsson9, 2021-08-04:
Interesting, but yes, after flashing you should always do a power cycle of the controller since there is one more microcontroller on the board which also needs to be rebooted.

But ok, I have made the necessary changes to make this respect wheel size, will test it on my bike and then do a commit to fix this issue.

As for push walk, yes, that is what I'm guessing, push walk speed limiting may need some additional logic to make it perfect.

fpdragon, 2021-08-04:
Maybe you could help me with an other problem:

I no wanted to configure the different speed limits for the different pas levels in the source code. I did the following:
```
	uint8_t current_limits_arr[9] = { 10, 20, 30, 40, 50, 60, 80, 90, 100 };
	uint8_t max_speed_percent_arr[9] = { 40, 45, 60, 65, 70, 77, 85, 93, 100 };
	for (uint8_t i = 0; i < 9; ++i)
	{
		g_config.assist_levels[0][i + 1].flags = ASSIST_FLAG_PAS | ASSIST_FLAG_THROTTLE;
		g_config.assist_levels[0][i + 1].target_current_percent = current_limits_arr[i];
		//g_config.assist_levels[0][i + 1].max_speed_percent = 100;
		g_config.assist_levels[0][i + 1].max_speed_percent = max_speed_percent_arr[i];
		g_config.assist_levels[0][i + 1].max_throttle_current_percent = 100;
	}
```

Now I am getting a linker error:
`?ASlink-Error-Could not get 11 consecutive bytes in internal RAM for area OSEG.`

Do you have made any memory map that needs to be considered?

danielnilsson9, 2021-08-04:
Ahh you hit the limited stack space! Constant troubles...

I see I have made a mistake, there should have been an __xdata in front of current_limits_arr.

Like this:
__xdata uint8_t current_limits_arr[9] = { 25, 34, 43, 51, 60, 68, 74, 82, 90 };
__xdata uint8_t speed_limits_arr[9] = { 10, 15, 25, 30, 35, 35, 35, 35, 35 };

If you want to configure speed limits per level in kph, just set global max speed to 100.

fpdragon, 2021-08-04:
> __xdata

perfect :-)

Now I am at the point where I can say that I don't need the HW mod any more.
Only cosmetic things need to be fixed now like the soft kick in and maybe the battery indicator.

Btw. the pattery indicator bar at the display is in the middle range when the battery is fully charged. Is this meter value also from the controller's firmware or is it a display thing?
I guess it must have to with the firmware since the original firmware has allways shown 100% and now I am at about 50% although I am using a higher voltage than originally designed.

danielnilsson9, 2021-08-04:
Was out testing my fixes, seems to work at least but not perfectly smooth. A warning about global speed limit though, what you configure in the display will override what you configure in the firmware if you have a display connected.

I will not implement battery SOC estimation, if anyone wants to do that it is fine. I think it returns the voltage at the moment, so if it is showing 50% that would be 50V.

Can't you switch your display to show battery voltage instead which I find more useful? Then you get a decimal point too.
Or maybe it is not possible on the 500C?

fpdragon, 2021-08-04:
> Was out testing my fixes, seems to work at least but not perfectly smooth. A warning about global speed limit though, what you configure in the display will override what you configure in the [...]

It's not such a big deal but it would be much cooler if this would also work.

I was talking about the graphical battery bar. I configured the display to show voltage and it does it correctly. The only thing is the bar itself that is about in the middle and I have not noticed that it has moved. So maybe it is just uninitialized and the default value is that it is in the middle at 50%? Or maybe it takes the voltage value and interprets it as percentage. Then it would be about 60V -> 60% which would be also the middle.
So I was wondering if there is any specific telemetry that may gives the value for the battery bar. in the end it would be easy to calculate a rough battery percentage from the min and max voltage and the current voltage value.

fpdragon, 2021-08-04:
btw, do you have paypal or is there a way to directly get in contact with you?
I would like to spend you a beer or two for your awesome work ;-)

fpdragon, 2021-08-04:
shall I open a new ticket for the battery bar indicator?

fpdragon, 2021-08-04:
reopened until the new sw changes by daniel were tested

danielnilsson9, 2021-08-04:
> It's not such a big deal but it would be much cooler if this would also work.
>
> I was talking about the graphical battery bar. I configured the display to show voltage and it does it correctly. The [...]

I see, so the bars are always showing on this display?
Yes it's more likely 60V then.

As you say, it should be easy to do something only voltage based, the reason I didn't do it was that then you would be required to configure you battery size too, (13S, 14S, 15S etc.) but I guess one could live with that.

But yes, please open a new issue for that.

I send you a PM on endless sphere for my contact details. :)

I will close this for now since I have made my commit.

commit 8aa66160 (2021-08-04): fix #4, speed limmiting not working
closed 2021-08-04 by danielnilsson9