danielnilsson9/bbs-fw #6: Battery bar indicator is not showing the correct value on display (Display 500C)

fpdragon, 2021-08-04:
The display shows the correct voltage values but the graphical bar is always in the middle of the range.
This could be because a voltage of about 40V - 60V is interpreted as percentage.
Or the bar indicator is just uninitialized and just shows 50% on default.

It's currently unclear which OPCODE could set this telemetry and influences the bar value.
If this could be found out then it would be easy to calculate a rough battery percentage with min/max voltage and the current voltage reading.

danielnilsson9, 2021-08-04:
Only battery related read request sent by displays are OPCODE_BAFANG_DISPLAY_READ_BATTERY (0x11) which is a request to controller firmware to get battery percentage.

Implemented in extcom.c:
process_bafang_display_read_battery()

Currently returning battery voltage instead of percent.

Needs to add config param for battery voltage in order to perform simple linear battery percentage calculation or implement more advanced SOC estimation algorithm.

fpdragon, 2021-08-04:
> Only battery related read request sent by displays are OPCODE_BAFANG_DISPLAY_READ_BATTERY (0x11) which is a request to controller firmware to get battery percentage.
>
> Implemented in extcom.c: [...]

Ahhh I see :-)

Nice

fpdragon, 2021-08-04:
I hard coded the values for my 15s battery pack:

```
static int8_t process_bafang_display_read_battery()
{
	if (msg_len < 2)
	{
		return KEEP;
	}

	uint8_t value = motor_get_battery_voltage_x10() / 10;

	value = MAP(value, 45, 61, 0, 100);

	// should be in percent but can't be bottered to do SOC calculation.
	// return in volts instead, i.e. 57% on display will correspond to 57V.
	uart1_write(value);
	uart1_write(value); // checksum

	return 2;
}
```

It works perfectly well. If we add the max voltage to the config then this could be added to the repo.

Thank you Daniel!