danielnilsson9/bbs-fw #7: Shift sensor is not working or interrupts very shortly

fpdragon, 2021-08-04:
I drop this here since I am unsure about the behavior of the shift sensor.

On my bike, as far as I can say, the shift sensor interrupted much more than on the open source fw. I am even unsure if the shift sensor works at all.

The sensor leds are blinking very weakly in red and sometimes there comes also green light.

Is the shift sensor usage implemented in the firmware?
Is it on the same port as the brake sensors?
Was the code tested? Maybe it's just me...

Thank you.

danielnilsson9, 2021-08-04:
As far as I know the shift sensor is only a brake signal and is connected to the same pin as the brake.
I have found nothing to indicate otherwise but I don't have a shift sensor so I do not know to be honest.

It might be because of the missing slow current ramp up it appears to be not working but I have no idea.

fpdragon, 2021-08-04:
Hmmm... I don't think that the slow ramp up influences the brake sensors / shift sensor.

The brake sensors are working well. I have a symbol that indicates when braking and maybe it's a little bit delayed but it works.

If the shift sensor is on the same port as the brake sensors then I guess the same display symbol should light up when shifting (as when braking)

Maybe when the config tool works, then I could look into the diagnositc logs and find out if the shift sensor is detected.

danielnilsson9, 2021-08-04:
There is no code related to shift sensor in this firmware so you will not find anything in those logs. Does the motor stop at all when shifting?

danielnilsson9, 2021-08-04:
I might be wrong though, on controllers having a separate gear sensor cable it might be connected to another pin, if so then it is not implemented.

danielnilsson9, 2021-08-04:
I had a look in my microscope, this was an easy one. It is connected to P2.6 on the MCU. Should be straightforward to implement.

fpdragon, 2021-08-04:
> I had a look in my microscope, this was an easy one. It is connected to P2.6 on the MCU. Should be straightforward to implement.
> 
> Maybe you want to have a go on this since I cannot test it properly? You can add it to sensors.c/sensors.h in the same way as brake sensor is done and then use it in app.c to pause motor when gear sensor is activated and send a pull request when done. It is most likely active low, just as the brake, i.e. when signal is low the sensor is active.

You are the man! :-)
I would really love to send you a beer or a brake sensor or something for your effort but I am having problems to get in contact with you (non public).

I will try to do the mod in the source code and come back to you.

fpdragon, 2021-08-04:
Is there also an own op code for the shift sensor to be sent to the display?

danielnilsson9, 2021-08-04:
I sent you a PM on endless sphere with my email address earlier today, check your inbox :)

There might be an OPCODE for that, do you have a separate thing on your display that usually lights up when shifting gear?

If so it is likely one of OPCODE_BAFANG_DISPLAY_READ_UNKNOWN1, OPCODE_BAFANG_DISPLAY_READ_UNKNOWN2 or OPCODE_BAFANG_DISPLAY_READ_UNKNOWN3 in extcom.c. I was never able to identify what those did when sniffing communication.

fpdragon, 2021-08-04:
Shift sensor now works for me.
P2.6 is the way to go. 
As you said, I added an own function for the shift sensor.
Since the shift sensor shall not reset the current ramp up, I had to restructure the `app_process()` function a little bit.

Here are my code fragments:
pins.h
`#define PIN_SHIFT_SENSOR				2, 6`

sensors.h
`bool shift_sensor_is_activated();`

sensors.c
```
bool shift_sensor_is_activated()
{
	return !GET_PIN_STATE(PIN_SHIFT_SENSOR);
}
```

app.c
```
void app_process()
{
	uint8_t target_current = 0;

	if (assist_level == ASSIST_PUSH)
	{
		if (g_config.use_push_walk)
		{
			target_current = 10;
			motor_set_target_speed(40);
		}		
	}
	else
	{
		// never limit motor rotation speed
		motor_set_target_speed(0xff);

		uint8_t throttle = throttle_read();

		apply_pas(&target_current);
		apply_cruise(&target_current, throttle);
		apply_throttle(&target_current, throttle);	
	}

	apply_speed_limit(&target_current);
	apply_thermal_limit(&target_current);

	apply_ramp_up_limit(&target_current);

	motor_set_target_current(target_current);

	/* enable / disable the motor */
	if (false != brake_is_activated())
	{
		/* reset the current ramp up */
		last_target_current = 0;
		/* disable motor */
		motor_disable();
	}
	else if (false != shift_sensor_is_activated())
	{
		/* disable motor */
		motor_disable();
	}
	else if (0 < target_current)
	{
		/* reset the current ramp up */
		last_target_current = 0;
		/* disable motor */
		motor_disable();
	}
	else
	{
		/* enable motor */
		motor_enable();
	}
}
```

danielnilsson9, 2021-08-04:
Nice!

I realized it does probably not have its own OPCODE if reported. Instead it would most likely be part of the status code where errors and states are reported to display.

Currently implemented ones in this firmware are (app_get_status_code):

#define STATUS_IDLE						0x00
#define STATUS_PEDALING					0x01
#define STATUS_BRAKING					0x03

#define STATUS_ERROR_THROTTLE			0x04
#define STATUS_ERROR_LVC				0x06
#define STATUS_ERROR_HALL_SENSOR		0x08
#define STATUS_ERROR_PHASE_LINE			0x09
#define STATUS_ERROR_OVER_TEMP			0x10
#define STATUS_ERROR_CURRENT_SENSE		0x12

fpdragon, 2021-08-04:
> Nice!
>
> I realized it does probably not have its own OPCODE if reported. Instead it would most likely be part of the status code where errors and states are reported to display.
>
> Currently [...]

I think the original firmware has sent it as STATUS_BRAKING since my display was also showing the brake symbol when shifting. I guess that's good enough

fpdragon, 2021-08-05:
I think this has to be reopened.

The shift sensor need a delay after it was detected since the shifting itself needs some time. ( I will go with 700ms for now).

The other thing is that it seems that the motor interruption takes way to long after the shift sensor is detected. So the shifting is still unclean and under heavy motor load allthough the motor current is interrupted afterwards.
So I guess we have to speed up things a little bit here.

danielnilsson9, 2021-08-05:
I see. Not sure how interruption could take long though, should be instant when signal is detected. 
Maybe has something to do with your loop seeming to be running very slow? Try disconnecting the display and run without it and see if that makes any difference.

fpdragon, 2021-08-05:
> I see. Not sure how interruption could take long though, should be instant when signal is detected.
> Maybe has something to do with your loop seeming to be running very slow? Try disconnecting the display and run without it and see if that makes any difference.

What do you mean with "my loop". I have not added any additional loop. All my code should not affect the timing of the code.

danielnilsson9, 2021-08-05:
You said earlier today that your average main loop execution time could be as high as 30ms when it really should be 1ms which I have measured. We talked about this might having something to do with that you have a different display than I have.

I just tested this by shorting gear sensor pin to ground and the motor loses power immediately. So something is wrong if yours do not behave this way and we need to figure out why.

fpdragon, 2021-08-05:
> You said earlier today that your average main loop execution time could be as high as 30ms when it really should be 1ms which I have measured. We talked about this might having something to do with [...]

I agree.

danielnilsson9, 2022-07-27:
I will close this as I am unable to test it. In theory the current ramp up should help with this, will await feedback from release.

closed 2022-07-27 by danielnilsson9

cdbrow, 2022-10-30:
This is still an issue on 2 of my bbshd builds. There needs to be a longer delay so the motor doesn’t kick back in while the shift is occurring.