danielnilsson9/bbs-fw #8: Undervoltage protection starts to toggle when getting out of range

fpdragon, 2021-08-05:
This may not be the most important thing but I sill want to report it to be tracket.

I am talking about:
`	g_config.low_cut_off_V = 45;`

First, under high load it can be that the undervoltage protection gets active much later and the voltage drops significantly lower than the protection limit is set. But that could be because of my weaker battery and the high load which results to a fast voltage drop.

However if the undervoltage gets active it blocks the motor power for a few seconds and then it gets active again since the voltage comes up again. This results in a kind of motor on, off, on, off stuttering which is irritating and seems to be currently the subtile signal for turning the motor off.

Two solutions came in my mind to solve this:

1st and more easy:
Just turn off and never turn on again if the low cut off is reached. This is the behavior I have seen on other ebike controllers. So even when the battery voltage recovers to a level above the undervoltage protection level, it wont turn on again and keeps in pas 0 and ignores throttle and keeps the motor disables.
If the user want so, he or she could power cycle the bike and then the motor is active again. However it does not make to much sense to bypass this because the battery is stressed and it is expected that the low cut off will activate again in short time. The only thing is that the user may wants to switch to a lower pas level and go for some more meters.

2nd solution which is more elegant:
It would be much better digital control that monitors the voltage and tries to control the motor current (e.g. with `target_current`) to be reduced to fit the low cut off voltage.
This should result in a natural behavior where the power of the motor gets weaker over last minutes until it turns completely off.
It would be much less stressfull for the batteries.
The user could go on driving and be sure that he squeeses out the battery to the exact point that is configured with the low cut off voltage. So no power cycle needed. 
The implementation could be similiar to my non official ramp up hack. There must be a member variable that holds voltage limited `target_current`. Every time the undervoltage protection is underreached this value may be reduced by 1% from the main loop until it gets 0. If the voltage recovers and goes up again this value could be raised by 1% again but that is optional. If the control works correctly then it should not be necessary to rise this value again since the battery is drained very softly to its end voltage.

Maybe I will prototype the second solution if I have time and drop the code here.

danielnilsson9, 2021-08-05:
Low voltage protection is implemented in NEC MCU firmware and not part of this code really.
But I guess we could probably do your first idea but I'm not sure if we should. Having the motor turned off/on is a good sign you have hit the lvc and not some other failure (if you don't have the display, otherwise you will see it there of course).

And for the second suggestion, that doesn't sound very healthy. I think most of us set lvc as low as we can tolerate but we make sure to almost never hit it as it is bad for the battery to be discharged that low. Eeking out even more power and draining the battery further by lowering current automatically would be bad idea then.

If you want something like that then it should probably be a power saving mode which is hit way before lvc is hit, e.g. when 20% remains (like on some smartphones).

fpdragon, 2021-08-05:
Having no display is a good point I don't have thought of.

However my argument is still valid that it is not good for the battery to switch off/on at full load. That's a horror scenario for a low capacity pack and it brings nearly no additional advantage to get this short motor pushes while riding.

I see that the low voltage protection is a HW feature.
What I was talking about with solution 2 is implementing a soft solution that just tracks the current voltage (that we know) and limit down the` target_current` value.

casainho, 2021-08-05:
On TSDZ2 our OpenSource firmware, as soon the lvc is hit, the motor current decreases a step so if  current is big, the faster thr current will drop but if you ride slowly, you still get power assist until it finally stops to gove the assistance.

And we also automatically calculate the battery resistance and calculate the battery voltage drop. Also with this drop we calculate and show to user the total power used: motor power + battery power loss. At high currents, battery power loss can be something like 25%!!
With this total power, we also estimate with good accuracy the battery SOC.

So, it is important for user to see his battery resistance / power loss, as the higher the value, the faster to hit the lvc.

fpdragon, 2021-08-05:
I think something similar to the TSDZ2 and as I described before makes sense.
It protects the battery further against high load voltage drops and it gives to opportunity to squeeze out some more watts without risking damage and at a lower current.

I am currently working on a system_ms() solution for the power ramp up hack. If I am finished with that then I should be able to implement this additional undervoltage control. I will let you know when I have something to show you.

danielnilsson9, 2021-08-05:
Alright, sounds good, looking forward to it!

fpdragon, 2021-08-05:
Ok, I finished my coding and I already flashed it to my bike.

But I have two problems now:
1.) Currently the weather is very bad here in Austria so I can't drain my battery in the mountains to test this code.
2.) I am still having problems in merging my changes with yours. All your code changes from the last day are not included right now. Each code (yours and mine) seem to work fine on their own. However, after I merged the code together I get the typical linker error regarding stack/ram and I am not sure where to search and where the most critical stack usage appears.

What a mess... Somehow we have to solve this otherwise we will get two permanent forks which would be too bad.

Daniel, I have commited everything in my fork repo. If you have any idea where we could do further optimization regarding RAM, it would be appreciated.

fpdragon, 2021-08-06:
software is implemented and tested successfully.
can be found on my fork in the branch "`f_Lays_StableBranch`"

closed 2021-08-06 by fpdragon