danielnilsson9/bbs-fw #10: Latest changes (ADC module?) destabilizes software?

fpdragon, 2021-08-06:
I have good and bad news and the bad first:

I tried to make a tour around on the nearby hills and wanted to test the big merge full blown latest software. After a few km I noticed that something was wrong.
The motor power drops sporadically for a few seconds and then kicks in with full power. Kind of the off/on behavior we had before the ramp up code but the trigger is sporadically. Latest ramp up code seems to be not active at these power drops.
Additionally, I noticed that the voltage seems to have also wrong readings some times. I conclude this because I use the voltage value to calculate the display’s battery bar in percent. This bar sometimes jumps to max (which can’t be from an electrical standpoint).
Both problems are new and I have not seen it on my last ride.
First, I tried to debug on the go. I disconnected the shift sensor and the brake sensors. Still the same problem. My read is that it maybe is no good idea to disconnect the shift sensor if the pin is read since it seems to electrically float around. Btw, maybe we need a setting in your app to enable/disable the shift sensor. Anyhow.

At home I tried to find the cause of the problems by comparing and testing the latest code changes themselves. I started with the code base I have used on my last ride.
Speed Sensor limit still works
Ramp Up with timer works
Soft Undervoltage works
Brake Sensor works
Shift Sensor works (kind of, there is still the known problem with the delay)

Then I merged into the latest code base from the master branch and the problem starts.

So, there are still many potential problem causes and I don’t want to claim that the latest changes on the master branch are unstable. Maybe I have done something wrong when I tried to merge the code and I will look into it again. However, I think we have to consider that some of the latest master branch changes destabilized the software completely.
Maybe the ADC module introduction still has a bug? Or something else I have not looked into.

Daniel, if you find out something then please let me know and I will try to merge again to put together our two worlds. 😉

The good:
In my fork repo I created a branch called “`f_Lays_StableBranch`” and here I merged into all features I have implemented in the last days. This branch is tested on a ride of 50km and seems to be stable as far as I see. It is ready to be merged with a (stable) master branch code base.
And if you allow me, the bike feels awesome now with all these soft and natural behavior.
Thank you Daniel, for making this possible!

danielnilsson9, 2021-08-06:
There could be issues introduced in the changes done recently because I have not tested these on a bike, only on my motor on the desktop where it seemed to work but I suspect there could be some new bugs there...

Have you had any success in confirming the frequency of you main loop? If there are issues it may break the rewritten serial communication I did in motor.c which probably causes those issues then. But I won't have time to work on this any more this week.

When you talk about ADC reading of battery, do you use the reading from motor_get_battery_voltage right? That is not an adc reading done on this chip, that is implemented in the motor control microcontroller and received through a serial interface.

But I think that measurement is also connected to some pin on this microcontroller and we could do a separata ADC reading but I doubt it would be any better due to all issues with adc that I have run into.

fpdragon, 2021-08-06:
> There could be issues introduced in the changes done recently because I have not tested these on a bike, only on my motor on the desktop where it seemed to work but I suspect there could be some new bugs there...

I guess it’s no surprise that you haven’t seen this on a motor on the desktop. Can you test it with a real bike?

> Have you had any success in confirming the frequency of you main loop? If there are issues it may break the rewritten serial communication I did in motor.c which probably causes those issues then. But I won't have time to work on this any more this week.

Sorry not now. What I can say is that your latest changes seem to behave normally first. But when riding I have these abrupt power drops from time to time (maybe 1 to 5 per minute).
I have not seen any errors on the display and beside the battery bar issue, everything seems to act normal. I guess serial communication errors would lead to an error 30h?
I have no oscilloscope and currently have not opened my motor at all. It is still well sealed. So I am not sure how to measure the loop speed.
Maybe I could build a test software and use the battery bar indicator to display a number from 0 to 100 and somehow code my measurements in there. Sorry, my time is limited for that. Any idea how to easily measure this? I have no better idea than using the execution_periode_ms value and somehow print it out over the battery bar. Maybe a tenth of this value then I could cover one second. 
 
> When you talk about ADC reading of battery, do you use the reading from motor_get_battery_voltage right? That is not an adc reading done on this chip, that is implemented in the motor control microcontroller and received through a serial interface.
 
Maybe I wasn't clear. I haven't expected the ADC module to be in a direct relationship with the battery voltage measurement. Currently I don't really know since I haven’t worked into this on that level.
I just can say that I have faced these two problems and I have seen that one of the larger changes was the introduction of the ADC module. I can rename the ticket if it comes out that this module works fine. 

> But I think that measurement is also connected to some pin on this microcontroller and we could do a separate ADC reading but I doubt it would be any better due to all issues with adc that I have run into.

Sounds to be a hard one. I hope I can support you with that.
At least from my point of view we already have a very good and practical variant, although many things seem to be still not understood and could be improved.
Have you ever tried my branch on a bike? Would be also interesting the other way around, if my code works for your platform.

danielnilsson9, 2021-08-06:
Ah I see, you are referring to the adc.c/adc.h files? That code was just moved in there from other placed and refactored a little, but the it only handles reading throttle position and controller temperature, not battery voltage, so I highly doubt it would have anything to do with your issues.

fpdragon, 2021-08-06:
Could be that the first problem has nothing to do with the battery voltage. I can't say.
The problematic motor power interrupts at every speed and at every battery level.
If I wouldn't know it better then I would say that this must be a defective electrical contact but we know it better and the other build works fine. So it can't be a hw probem, I would say.

fpdragon, 2021-08-06:
I have done my work and build a debug test program for loop measurement.

As said, I use the display's battery bar to indicate a number between 0 and 100.
However, it's not really that I see 100 states since the indicator bar has just 10 blocks. But at least I see if the value is between 0, 10, 20, 30, ... and so on.

I put out the variable `execution_periode_ms`.
First, I ran the code with `execution_periode_ms/10`. The first bar would be 100ms and the highest >=1000ms.
The indicator showed no bar what is good since we are somewhere between 0ms and 100ms.

Fine, in the next step I directly wrote `execution_periode_ms`. The first bar would be 10ms and the highest >=100ms.

I see two different situations:

1st is the common with two bars (10ms - 20ms). 
This seems to be the standard case when the main loop does just standard stuff.

2nd situation comes in sporadic pulses and is showing 8 bars (70ms - 80ms).
These pulses are seen in intervals of about 1 to 5 seconds and the value is shown for a moment.

Maybe the values are incorrect and we also have to add another 10ms each. Depends on how the display represents the values. But for now, I guess it's not so important to know exactly 70ms to 90ms is still a value we can work with.

I must say that I am surprised that these measurements are fitting nearly exactly to our calculations we had before from the first ramp up implementation. However, this does not help us finding the cause.
If you can definitely say that the main loop runs way faster on your machine, then this seems to be a miracle. I hope we can find out what slows my device down and we find a solution.

In the end I can say that the shifting can’t run cleanly with a 20ms or even a 80ms delay. This explains it well. A good shifting interrupt should start in a part of a ms and should be held for several 100ms to be compleded cleanly.
I think you know the processor better than I do. Is it possible use a HW interrupt for the shifting sensor pin to speed up things? If yes, how does a HW interrupt would fit into your execution design?

danielnilsson9, 2021-08-06:
Thank for doing this investigation!

On what version did you perform these tests? My master or your stable?
If it is on your stable then the sporadic big jumps are expected so no worries there.

Slow time is very odd... Any chance you could also test this without the display connected? Use config tool from my master branch and print the loop time every e.g. every 100th loop to the event log in config tool.

Not possible to use external interrupts, P2.6 does not support it. But will not be a problem when firmware runs as expected. We can remove the delay and run the loop at 1kHz...

danielnilsson9, 2021-08-06:
Please test this in my master branch, go to main.c and uncomment "#define DEBUG_LOOP_TIME_EVENTLOG".
Connect config tool and check the eventlog tab after flashing.

fpdragon, 2021-08-06:
Currently I have tested with my branch "`f_DebugCode`" which is a instrumentation of "`f_Lays_StableBranch`". Btw, sorry for the names but I am a little bit lost without strict coding and work standards. xD

Maybe I have time to retest with your `master `branch in a few hours.
I would merge my debug code into the `main `branch and retest it in that way.

The problem with DEBUG_LOOP_TIME_EVENTLOG is still that I can't use the config tool. Still the problem with the communication timeout.

danielnilsson9, 2021-08-06:
I have fixed config tool, you need to use the master branch for that. But I just realized what is causing this, as you said your are on the very old repo before adc.c/.h was introduced, the old throttle reading code had some experimental debug stuff and is really slow so thats is why your main loop runs at 40Hz...

But that is good, at least we know why! (I tested your stable branch, runs at 40Hz on my scope).

danielnilsson9, 2021-08-06:
Anyhow, there is most likely something wrong with the changes I introduced in motor.c then if as you said you applied your changes on top of my master and it didn't work very well. I will look into that.

fpdragon, 2021-08-06:
> Anyhow, there is most likely something wrong with the changes I introduced in motor.c then if as you said you applied your changes on top of my master and it didn't work very well. I will look into that.

Thank you.

I will have a look about the config tool and if it works now.

danielnilsson9, 2021-08-06:
Hahah took a ride on master, it's absolutely crazy now, a real death trap! 😄

fpdragon, 2021-08-06:
> Hahah took a ride on master, it's absolutely crazy now, a real death trap! 😄
> I will have to roll back commit for commit and see when it started going wrong...
> 
> EDIT: reverted changes in motor.c, seemed to work normal after that but only tested with throttle.

Haha... Yes it was quite "uncomfortable" but I guess I have a lower current limit than you (15A) so I was able to stay in the saddle.
Good to hear that you were able to reproduce this problem.

fpdragon, 2021-08-06:
with which branch shall I dump the event log? With the `master `from 06.08.2021 21:03:50 or with `motor_async_io` ?
Does the new stable master already have the relevant main loop timing updates that are of interest or shall I go to the unstable one just for measure?

danielnilsson9, 2021-08-06:
You can take the latest master, but I'm sure you will se 4ms and maybe some sporadic 60ms! But can be good if you verify that config tool works with the bafang programming cable, so go ahead

fpdragon, 2021-08-06:
Let's see... 45A  with my 60V battery pack... would make... holy shit... 2700W xD
When it gets to these power levels I stick with my good old 120PS kawasaki motorcycle.
BTW, from 08.08.2021 on I am on a motorcycle tour so I won't be able to develop about a week long.
But I will be back ;-)

fpdragon, 2021-08-07:
I flashed the latest master branch and I confirm that now the config tool is working. :-)
I see the debug messages and about each 1.5 seconds I get a printout that says that the main loop takes 4ms.

fpdragon, 2021-08-07:
And... as you expected, sometimes 66ms

fpdragon, 2021-08-08:
I merged the latest master again with my stable branch and now I go no motor support at all.
See my branch "f_Lays_MergeBranch".
I will stick to my branch "f_Lays_StableBranch" for now.

dav0000000, 2021-11-04:
Have found the async motor.c changes to work very reliably and will no longer use the sync version.

I also saw the jump to 64ms in the sync version. I managed to get it down to 25ms by only letting motor_process run 1 of the 4 updates per loop using else if. But then I switched back to the async motor.c and the loop is 100% reliable in my opinion. I have done quite a bit of riding and it runs flawlessly.

danielnilsson9, 2021-11-04:
Alright! Sounds good, thank you for testing! I have run it quite a bit also and not had any issues so I will merge that branch to master now. Have you been running anything with a throttle, if so, noticed any issues?

dav0000000, 2021-11-04:
I don't have the throttle connected.

danielnilsson9, 2022-07-27:
Fixed.

closed 2022-07-27 by danielnilsson9