danielnilsson9/bbs-fw #11: Speed on display sometimes has unrealistic jumps to high values

fpdragon, 2021-08-06:
Not sure what we can do here but I just drop it here for now to get together and find new ideas.

The problem is that sometimes (at least) my display (500C) shows high-speed spike readings for short moments that are totally wrong. This also can happen when the bike stands still and the wheel is in an unfavorable angle (magnet aligned to the sensor). 

As far as I know, the speed sensor is just a hall sensor that reads the magnetic field and detects a permanent magnet when the magnet rotates nearby on a spoke. Then an electrical pulse is sent to the controller.

Maybe it would be possible to improve this by doing a kind of debouncing? We do know that extremely fast pulses are unrealistic since the wheel is expected to rotate at a naturally limited rotation speed.
Just an idea...
What do you think?

2nd extra:
Additionally, we could improve the timeout behavior of the speed sensing. When riding fast and stopping abruptly the controller thinks that we are currently on the measured speed and waits for the next speed sensor pulses. Since the wheel has stopped, there are no more pulses and the controller has to run in a timeout until he decides that we must have stopped. This timeout has to be relatively high to measure lower speeds. In this timeout time he still thinks that we are riding fast.
This could be improved. When the waiting time rises then the controller could know at least that the speed must be lower than the speed before. And as the timeout progresses the controller could use lower and lower values until it reaches 0/stop.

danielnilsson9, 2021-08-06:
1
I have never observed that but I haven't done too much riding with the display. I will try to see if I can reproduce that someday.
And yes, we will have to debounce if that is what is happening, the check runs in a interrupt on a timer which executes every 100us, see "sensors.c".

2
Yes the timeout might be a bit high as you say. I think it is set to 5 seconds, see sensors.c in interrupt routine.
It was set pretty height to be able to measure really slow speeds but could probably be tuned and as you say could benefit from being adjusted dynamically. Please experiment with that if you want! :)

dav0000000, 2021-11-04:
I happy with the timeout after changing it to 2.5 seconds. I was also experiencing high jumps and some resets to 0. It looks like it may be fixed by ignoring the first speed sensor count as in the code below in sensors.c.

static __xdata bool pas_sensor_discard_first;

void sensors_init()
{
	pas_sensor_discard_first = false;
}

INTERRUPT(isr_timer4, IRQ_TIMER4)
{
	// speed sensor
	{
		bool spd = GET_PIN_STATE(PIN_SPEED_SENSOR);

		if (spd && !speed_prev_state)
		{
			if (speed_sensor_discard_first)
			{
				if (speed_period_counter > 0)
				{
					speed_ticks_minute_x10 = 6000000UL / speed_period_counter;
					speed_period_counter = 0;
				}
			}
			else
			{
				speed_sensor_discard_first = true;
				speed_period_counter = 0;
			}
		}
		else
		{
			++speed_period_counter;

			if (speed_ticks_minute_x10 > 0 && speed_period_counter > 25000)
			{
				speed_sensor_discard_first = false;
				speed_ticks_minute_x10 = 0;
			}

		}
		speed_prev_state = spd;
	}	
}

closed 2022-07-27 by danielnilsson9

casainho, 2022-07-27:
Congratulations in solving this issue, fue yo incorrect code on interrupt

commit aefd68d1 (2022-07-27): Fix error in road speed and pas cadence computaton. fixes #11