danielnilsson9/bbs-fw #20: Wish: outlining differences to original firmware

tefod-zz, 2022-04-13:
Hi Daniel,
could you outline in the docu/wiki the differences to the original firmware?
In addition I read on endless sphere somewhere, that some things (as like speed regulation) is not working.

Would great for new users, if you mention the differences and special the things from the pipeline, which are not ready yet.

Thanks :)

CitizenDev, 2022-07-18:
> Hi Daniel, could you outline in the docu/wiki the differences to the original firmware? In addition I read on endless sphere somewhere, that some things (as like speed regulation) is not working. [...]

Unfortunately this project seems abandoned.
I just recently did BBS02 750W conversion and was super happy when I found this github. Then read all posts on ES forum just to find out that currently compatibility with BBS02 is unknown and there are some issues with this iteration of the custom firmware related to throttle that the developer was not able to overcome and therefore quit.

For me personally the biggest benefit would be 2 different modes for assist (Legal/Off road setting) without this my bike is illegal to drive on the roads and I'm risking hefty fines every day I ride. This is so important to me, that if this thing is ever finished I would be easily willing to spend 50$ for it.

I'm yet to find another viable solution other than switching my brand new 850C display unit to Eggrider that costs another 100$ and has way inferior display unit itself with very small non-color display.

Last update on ES forum from danielnilsson9 at January 2021:
_I'm still stuck on the weird sporadic throttle readings and have grew kind of tired of this and is not really actively working on it anymore. I have been unable to implement any good filtering as the stack space on this microcontroller is so ridiculously small (128 bytes) that I can basically do no more calculations without heavily optimizing the rest of the firmware. Setting medium or large memory model in SDCC seem to produce non working code, or I have not found the correct parameters.

I have been going through a disassembly of the original firmware and have obviously found the throttle reading code and have implemented a version which matches it closely but to no help. So my best guess would be that they had the same issue in the original firmware and managed to filter it out somehow. My 8051 assembly reading skills are too bad though to follow the additional processing of the sampled value which may be present... anyone able and willing to give it a go?_

Zensaition, 2022-09-20:
I feel he took a bit outta of the original software in terms of ebrake stuff I found so far.

@danielnilsson9

closed 2023-08-02 by danielnilsson9