danielnilsson9/bbs-fw #23: BBS02 compatibility

CitizenDev, 2022-07-18:
Whats the current consensus regarding this firmware on newer BBS02 models?
Based on some information their controllers are nearly if not identical with probably lower power settings preset.

I'm willing to help test this if I can find original firmware for my BBS02 so I have something to revert to.

danielnilsson9, 2022-07-24:
I have no idea, it has not been tested as far as I know. From photos the PCB looks very similar and likely uses the same MCU:s so I would be very surprised if supporting this board would require a lot of work.

But as always, not recommended to try to flash this since you risk bricking your controller if you cannot get hold of the original BBS02 firmware.

Here is someone who seems to have access to the file but refuses to share it.
https://electricbikereview.com/forums/threads/error-7-bbs02-help-needed-brand-new-bike.36589/

danielnilsson9, 2022-07-24:
I found some high res pictures of the BBS02B controller here.
https://electricbike-blog.com/2016/07/12/bafang-completely-redesigns-the-bbs02-and-tells-no-one/

Although it looks very similar it is definitely not identical to the BBSHD controller and it is quite likely that some pins used on the MCU:s may not be the same making this firmware incompatible.

The PCB of the BBS02B controller would need to be traced just as I did with the BBSHD controller PCB to have a shot at supporting that board. I would need a BBS02B controller to do that, which I do not have access to.

danielnilsson9, 2022-07-26:
I have found the original 52V firmware for some version of the BBS02 here:
https://support.biktrix.com/kb/guide/en/52v-programming-for-bbs02-motor-K2gUq2nWAX/Steps/1132765

It seems to use the IAP15W4K61S4 MCU which is the same MCU that BBSHD version 1.5 uses.

I would be careful flashing this tough as it seems to have bricked many users controllers (probably did not have the correct version).

CitizenDev, 2022-07-29:
I finally got the cable and did MCU check with SPC Tool.
It's a very recent purchase and MCU check came back with:

  MCU type: STC15F2K60S2
  F/W version: 7.2.5S

I'll definitely donate something towards you getting access to BBS02 controller. 
Thinking about it I would maybe willing to pay 50% of the controller cost. But in case I happen to somehow burn mine in the future you would send it to me as a replacement. Like an insurance of sorts. I don't live far (Estonia) so shipping shouldn't be an issue.

danielnilsson9, 2022-07-29:
Interesting, that is yet another MCU model but it is in the same family so I'm not expecting much trouble.

The controller will likely be destroyed in the process, I have destroyed two perfectly fine BBSHD controllers during the development of this project...

At least it will be depotted in order to be able to put the PCB under the microscope to trace all signals. So it is not really in a good state to put back into the housing and motor after this.

I'm sure if the project gain some traction and people interested in BBS02 support chime in a few dollars we will get there in no time.

I doubt they would replay. I don't have much need for it a tleast.

CitizenDev, 2022-07-29:
No problem, I'll send you something nontheless and maybe preach in some forums to crowdsource at least one controller.

CitizenDev, 2022-07-30:
Additional note;
Based on serial number stamped on my controller (2203010073) it's fairly recent one (01.03.22) assuming 220301 is date and 0073 is unit production count. 
MCU might have changed within recent years because of shortage.

closed 2022-09-08 by danielnilsson9