danielnilsson9/bbs-fw #29: Implement proper battery SOC calculation

danielnilsson9, 2022-07-31:
And make it actually work, not like in the original firmware.

Should be possible now when the out of SRAM situation has been resolved.

dav0000000, 2022-08-13:
Does the SOC value need to be stored in EEPROM. I suspect so. I'm sure you know that there is a 100,000 EEPROM erases recommendation. Is there a UART code triggered when the display is turned off? That would be ideal if it is usable. If not then determining when to save the counter to EEPROM could cause problems with EEPROM failures down the track. "On-chip 16K/32K/40K/48K/56K/58K/61K/63.5K FLASH program memory with flexible ISP/IAP capability, can be repeatedly erased more than 100 thousand times."

danielnilsson9, 2022-08-15:
Yes, that is definitely something to consider. There is no UART message that can be relied on. Some bafang displays send PAS 0 selection when shutting off but some after market displays does not. I also suspect a lot of people just shut of their bike on the battery switch (which I do).

I have not decided if I will bother to implement coulomb counting yet as I'm not sure of the quality of the current reading sent from secondary controller. I will add a simple voltage based calculation in next release at least...

danielnilsson9, 2022-08-29:
Simple and incorrect linear voltage based (under no load) method implemented. I will not bother with coulomb counting. If anyone wants to investigate current measurement quality and contribute coulomb counting method, feel free to open a new issue to discuss the approach.

closed 2022-08-29 by danielnilsson9