danielnilsson9/bbs-fw #40: Option to show motor temperature in speed field

danielnilsson9, 2022-08-17:
Add an option to temporary show motor temperature in speed field while push walk is active.

danielnilsson9, 2022-09-03:
Implemented, display will show Celsius independently of unit setting since it goes to maximum 99 which would not be enough range for fahrenheit.

closed 2022-09-03 by danielnilsson9

CitizenDev, 2022-09-07:
Could this be changed to controller temp sensor for BBS02 since we dont have motor temp sensor?

danielnilsson9, 2022-09-07:
That is already implemented, if there are multiple sensors the one with the highest value will be shown. There is only one sensor on BBS02 so that will be shown.

silenec, 2022-09-09:
Would it be possible to have an option to also show other things in speed field such as amps, watts or even cadence? I don't really need speed because I see gps speed on my phone right below the bafang display and on my smart band. Might be useful for testing / adjusting parameters and finding the best motor / human efficiency.

danielnilsson9, 2022-09-09:
I will consider it, please create an issue so we can see if it is something more people would be interested in.