danielnilsson9/bbs-fw #46: Support motor core temperature sensor

danielnilsson9, 2022-08-30:
There is another thermistor on the hall sensor board for measuring motor winding temperature.
Add support for reading this sensor too.

Not present on BBS02, only BBSHD.

dav0000000, 2022-08-31:
In sensors.c line 211 you have:
adc_motor_x100 = adc_get_temperature_contr() * 100;

Should it be:
adc_motor_x100 = adc_get_temperature_motor() * 100;

dav0000000, 2022-08-31:
Are you getting a good value for motor temperature?
I just tested and was getting 21C at controller but 91C at the motor so I got thermal limiting and then motor temperature error. I disabled both of these. Then rode a little and the controller temp went up to 38C but the motor temp went down 1C to 90C. Perhaps I stuffed something up, I will check a bit more tomorrow by hardcoding a value in sensors for motor temp.

danielnilsson9, 2022-09-01:
Interesting, yes it is working for me, I get roughly the same value on my motor on the bench, 22 vs 23 degrees.
I haven't tried on my other motor on a bike yet though.

How old is your motor?

dav0000000, 2022-09-01:
Motor is about 18 months old. Controller is about 6 months old. I added "return 3300;" to temperature_motor_x100() in sensor.c and it shows as 33C in the app logs.

danielnilsson9, 2022-09-03:
Not sure what is going on, sound like you sensor in the motor might be broken. I have changed to configuration so that it is possible to select which sensors to use.

dav0000000, 2022-09-03:
Thank you. It is possible the sensor is broken as I do some very big hill climbing hitting controller temp 70C more often then I should. It is strange as yesterday up a hill I saw the motor temp reading go from 91 to to 87. Then it went back to 91 after just cruising again. It does seem like it's doing something with the temperature but in reverse. Again, thanks very much, your work, it really amazes me.

danielnilsson9, 2022-09-03:
Sounds almost like the pin might be floating, i.e. you have a bad connection, broken wire or faulty. It's an analog pin reading so if there is no connection you just get noise.

I have tested on my second BBSHD motor from 2020 and it is working there too, my other one is from 2017.

What is interesting is that Bafang might not use the sensor if you haven't had any issues in the original firmware?

danielnilsson9, 2022-09-03:
I stumbled upon some information while investigating something completely unrelated. So LUNA has developed their own BBSHD VESC based replacement controller and the source code has been upstreamed to the VESC project.

According to this code here and the comments, Bafang has switched to a PTC sensor instead of the NTC sensor which my code is based on. I'm fairly certain you have the PTC version which would explain why your reading seemed to drop when it got hotter.

https://github.com/vedderb/bldc/blob/master/hwconf/luna/hw_luna_bbshd.c

They have implemented some code for automatically detecting which sensor it is and adjusting the calculation accordingly.

dav0000000, 2022-09-03:
That is really interesting. I just flashed my wife's bike and also got the same 91 reading. I'll have a look at the link. Cheers.

dav0000000, 2022-09-04:
"return adc_get_temperature_motor()" comes back with 45, with controller temperature raising from 18C-28C it did not change. Really not sure the motor warmed up though as it was only a short ride but seems we are aiming to get a value between 930 and 1385. I will be riding again later and motor will definitely get warmed up. Is that value (45) workable in sensors.c or are changes also required to adc?

danielnilsson9, 2022-09-04:
Yes, I had to increase ADC resolution to 10bit in order to have sensible readings since the PTC thermistor is less sensitive (less change in resistance with temperature compared to the NTC thermistor).

I just pushed an implementation which I think is working, please try it out.

dav0000000, 2022-09-04:
Thanks.

With a cold bike, in the APP logs:
2022-09-04 13:01:03.142    Info    Temperature, motor=26C, controller=19C.
2022-09-04 13:01:08.136    Info    Temperature, motor=26C, controller=20C.
2022-09-04 13:01:13.554    Info    Temperature, motor=25C, controller=20C.

I then tested on a ride with only a small hill (less than 2 minutes up at about 25Amps):
motor=85C, controller=44C

I still have thermal limiting disabled for motor temp. It looks like the values may be a bit high.

danielnilsson9, 2022-09-04:
Yes that sounds a bit high. I'm looking at the table and comparing it to various data sheets of 1kOhm PTC sensors and it's definitely a bit off  from the standard tables which would cause a higher reported temperature. I believe LUNA made those measurements themselves, not sure what is going on. I'm thinking of going with a standard table and see if it produces more reasonable results.

Most seems to follow this PT1000 table:
https://www.sterlingsensors.co.uk/pt1000-resistance-table

danielnilsson9, 2022-09-04:
I pushed a new table, according to the new table you measurement was off by about 20 degrees at the top range which sounds about right (in my limited experience with running with a temperature sensor on the bbshd).

dav0000000, 2022-09-04:
After a very small ride I saw Controller = 27C and Motor = 30C. Looking good so far. I will test with some higher temperatures but maybe not till tomorrow as I'm hitting LVC now due to battery.

Maybe unrelated but I also saw some strange surges in speed up to 65km/h even though I was only doing about 25km/h on that short ride. Could that be due to the 2ms loop time or somehow related to the timer changes? I've never seen that happen before.

dav0000000, 2022-09-04:
Is the timer problem that timer4 is being disabled to get the values in sensors and that now it is timer 0?

	CLEAR_BIT(IE2, 6); // disable timer 4 interrupts
	SET_BIT(IE2, 6);

danielnilsson9, 2022-09-04:
Nice catch! Thank you, yes definitely that is the problem, totally forgot about that!

dav0000000, 2022-09-05:
Had a good ride today and can say the temperatures look good now with this new version for the PTC sensor. It's a few degrees higher after an overnight and it cools down slower than the controller like what would be expected. All good I would say. Thanks.

danielnilsson9, 2022-09-05:
Sounds good! None of them are calibrated so they won't be exact, but hopefully within a few degrees at least which is good enough. I haven't measured the beta value for the NTC thermistor in the controller so that one can be a bit of I suppose, will try to do that someday.

closed 2022-09-06 by danielnilsson9