danielnilsson9/bbs-fw #51: Sensors timer 0 causing instability with speed sensor

dav0000000, 2022-09-05:
After correcting the timer disable/enable code in sensors.c for TIMER0 I was still having instability issues. I would see the speed go up randomly sometimes as high as 99. After this, when I requested 500w power it would limit me back to around 250w. This could actually be due to the motor somehow resetting (if that is possible) and taking me from SPORT mode back to STANDARD which would actually explain the difference in power but I have only just thought about that so did not confirm.

I reverted sensors.c back to use TIMER4, added TIMER4 code back to stc15.h and removed the related sensor code in timers.c and timers.h and the problem has again disappeared. Was there a reason for changing sensors.c to use TIMER0. I thought I read somewhere that you had to use TIMER4 to check the state of the brakes, pas and speed sensors, but I have no idea where I read that.

danielnilsson9, 2022-09-05:
Strange, I have been doing quite a bit of riding and not observed this issue at all, will have to investigate more.
The timer change had to be made since the MCU present on the BBS02 controller only has 3 hardware timers where two must be used for baud rate generation on the serial communication ports. 

That left only one timer for other stuff which forced med combining the millisecond counter timer and the sensor polling timer into a single timer. In order to have the same code for BBSHD and BBS02 the change also applies to BBSHD.

dav0000000, 2022-09-05:
Ok well that is good. I'll try again when I get a chance and see if it repeats.

danielnilsson9, 2022-09-05:
Just took a 30km ride using pedal assist only with zero errors in speed reporting and zero unexpected power interrupts while pedaling. Before the change of interrupt disabling on timer0 I had roughly one noticeable power interrupt on PAS every km.

It makes me fairly certain that issue was resolved. Which makes me wonder what is causing your issues.

Two ideas:
- Unstable on 20MHz?
- Something in the PTC code path which you are using and not me

It would probably be helpful if you could run an unmodified version for the current code just to be sure it is still there.

dav0000000, 2022-09-05:
I ran with TIMER4 on 20MHz for 45km today without any issues so possibly not related to the frequency change or the PTC code directly. I better first test again if you are not experiencing any issues. I couldn't even get 100 meters without seeing the problem..

dav0000000, 2022-09-05:
Unfortunately I just tried again and got the same issues with a fresh, completely unmodified latest build. I only changed default_current_limits to 10, 15, 20, 30, 40, 55, 70, 85, 100 and g_config.max_battery_x100v = 5880 and took off in PAS3. It hesitated, then went, then hesitated, then went, then the speed went up to 99 then dropped off again, I turned around and went back after 20 meters. I did not do any further investigation. I flashed the version back with the TIMER4 changes and all ok again and went for a quick 5km ride. No issues whatsoever. Something is definitely not quite right. I think the enable/disable timers are causing some kind of problem but that's just the feeling I get as it seemed to get worse when we fixed the problem in sensors.c with TIMER4 being enabled/disabled instead of TIMER0. Do we need to disable/enable the timers to read these values? I did not actually see any issues with bad readings before you implemented that change some time ago. Shouldn't the volatile setting be enough on those variables?

dav0000000, 2022-09-05:
I should mention that I do not have a throttle, just PAS.

danielnilsson9, 2022-09-06:
That is crazy, what is going on. Which compiler version do you use?

Those variables has to be protected since they are 16bit and this is an 8bit controller, you don't have atomic read on a 16bit variable. So if an interrupt occurs when you had read the first byte the second byte may have changed when you continue the read later.

dav0000000, 2022-09-06:
Thanks for the explanation about atomic reads. Looks like my SDCC is v4.1.0.

danielnilsson9, 2022-09-06:
That's a difference at least, I have been building with 4.0.0. Will try to build with 4.1.0 tonight and see if I can reproduce.

dav0000000, 2022-09-06:
Sorry for the possibly 'inexperienced' questions, but is interrupt priority something that would change between TIMER0 and TIMER4 amd somehow cause this affect?

danielnilsson9, 2022-09-06:
No, no, it good to have some more eyes on this! 

But it looks like all interrupts have the same priority as I haven't configured any high priority interrupts and all are low priority by default. See page 252 in STC15W4K32S4-STCMCU.pdf in doc folder.

dav0000000, 2022-09-06:
Yes I've been looking at that doc. It was the info in 6.5 Interrupt Priorities on pages 266 and 267 that prompted my comment.

danielnilsson9, 2022-09-06:
Did a quick test with the 4.1 compiler and managed to reproduced it right away on my motor on the bench, there must be some bug in that version or some behaviour has changed. There are some warnings in the SDCC manual about overlaying when calling functions from interrupt service routines for non reentrant function, but it also states that it only affects the small memory model and I am using the large memory model... so I'm not sure... maybe something has changed there. Please test using SDCC 4.0.