danielnilsson9/bbs-fw #53: Add keep current option

danielnilsson9, 2022-09-07:
This option exist in the original firmware which decreases motor power on pedal assist when approaching assist level target cadence.
Configured in percent of assist level power to receive when at assist level target cadence.

Example:
PAS Target Cadence:  60% of 150rpm = 90rpm
PAS Target Current:   50% of 30A  = 15A
Keep Current: 70% of 15A = 10.5A

When pedaling at 90rpm the delivered current is 10.5A

Unsure about the mapping, linear relationship?

silenec, 2022-09-07:
Oh nice, my favorite feature. Actually the way I use this with the stock firmware is:

I have my PAS Target Cadence ("Speed" in original FW) unlimited / maximum on all PAS levels since I want to be able to pedal together with motor power at whatever cadence I want so that the speed that I reach is only limited by selected target current, power of my legs, wind resistance or hills. I also do not use any "Max Speed" (km/h) limits.

Could this be made to work independently of PAS Target Cadence? It still works great without any cadence limit with the stock firmware so I think that is how bafang implemented this. In your example, the keep current would do nothing with my PAS Target Cadence set to 100% / 150rpm because normally I do not reach a cadence so high.

Configurable PAS Target Cadence, Keep Current Cadence and Keep Current % would be perfect. 

I think Keep Current Cadence could stay the same for all PAS levels - the user just needs to select one cadence that is "too low" for him and when the cadence drops to this value, motor power increases and it tries to push you past this low cadence into more comfortable cadence. That way you need to change PAS levels less often. Once your cadence drops too low with stock firmware, it is just like if you selected the next higher PAS level. Except it works automagically :)

CitizenDev, 2022-09-08:
+1 to silenec implementation idea.

closed 2023-01-04 by danielnilsson9