danielnilsson9/bbs-fw #55: Power surging

Dweebur2, 2022-09-09:
Hello. I just installed the new firmware and tried it on a test drive. It appears my speed and amperage go up and down instead of saying steady. This happens with the throttle and with pedal assist. Please excuse me for I'm not very tech savvy. Is there a solution? Thank you in advance

danielnilsson9, 2022-09-09:
Strange, no other reports of that and nothing I have seen myself.
Are you on version 1.1? Which controller?

Dweebur2, 2022-09-09:
Hello .yes 1.1.0 version, bbs02, 750 w, 48 volt, 850c display. It is quite jumpy ramping up and down. I can see the wattage bar going up and down and feel power surges and then drops.  I truly appreciate all your help and time you've put into this.

CitizenDev, 2022-09-09:
> Hello .yes 1.1.0 version, bbs02, 750 w, 48 volt, 850c display. It is quite jumpy ramping up and down. I can see the wattage bar going up and down and feel power surges and then drops. I truly appreciate all your help and time you've put into this.

I have EXACT same combination of hardware. Can you share your config settings?

Dweebur2, 2022-09-09:
I basically used the default settings only changing the max current amperage to 18 and to show temperature during walk mode.  If you need more description I'd be glad to share it.

CitizenDev, 2022-09-09:
I cannot share my exact conf atm, but I think default it pretty aggressive.
Try current ramp 6-7A and configure PAS levels to increase current % and max speed incrementally and throttle end voltage at 4200 for more linear control.

Also please note that amperage should get lower when reaching speed or cadence limit

danielnilsson9, 2022-09-09:
I'm sorry I don't know what's going on with you controller. You can feel the motor power increase and decrease just when using PAS and no throttle? Sounds like some form of limiting is going on, please check that you temperature sensor reports normal values in the event log in config tool.

Dweebur2, 2022-09-09:
I seem to have a new problem after your firmware was acting kind of squirrely I reinstalled the original firmware. Now installing your firmware again I cannot seem to get your configuration tool to connect I've installed and reinstalled many times any idea?

dav0000000, 2022-09-09:
I had a similar issue. At close to 40km/h my bike was surging on the display even though the speed limit was much higher and I had plenty of power. It happened to me for a few days and made no sense. I have 46T front chainring and am only using the 24T rear cog as I'm waiting on some parts. I did the calcs and that chainring and rear cog result in the motor doing 160RPM at about 38km/h. Problem solved... needed to change gear.

danielnilsson9, 2022-09-10:
@Dweebur2 Did you remember to set frequency to 20MHz when you flashed my firmware again? 18.432MHz is used for bafang firmware, this needs 20

danielnilsson9, 2022-09-10:
Did this behaviour happening when you were hitting you configured speed limit or was it any speed?

Dweebur2, 2022-09-10:
Hello.... no I did not use 20 mhz.  that might be the problem right there.  I will try it again and give you my results.  This was happening at all speeds.

Dweebur2, 2022-09-11:
Well I went back through and followed instructions thoroughly. Still feels like it's accelerating and decelerating quite a bit even when speed limit is set to 100% and with throttle only. Its very usable just different than original firmware with the same settings.  I love what your doing and appreciate all your hard work, will make a donation.

danielnilsson9, 2022-09-11:
Definitely sounds like something is wrong because that is not expected behaviour. If there is some problem with the reading of you speed sensor that could explain it, one thing you could try is to uncheck the speed sensor box in the config tool and see if the behaviour changes. Also please check the event log in config tool to confirm that you temperature reading is sensible, otherwise you could also potentially get this behaviour you are experiencing.

danielnilsson9, 2022-10-03:
For anyone having this problem, please try 1.2.0 and let me know if it works or if the issue is still present.

aaron-manning, 2023-08-06:
I get surging, but only when using the speed / cadence limit features

BBS02, v1.3.1

Dweebur2, 2023-08-07:
Hey Aaron..thanks for your response.  Is this with your bbs02? I had my speed at 100% with current going up 10% each power assist level.  I'm going to give the newer software a try.  Any advice you could give me would be greatly appreciated.  Have you had success with your bbs02 and this newer firmware?

silenec, 2023-08-07:
BBS02 did not really work for me with PAS on v1.3.1. Throttle only or cruise was ok. Maybe your problems are also related to https://github.com/danielnilsson9/bbs-fw/issues/114 ?
I think you need to compile a firmware with all the recent fixes or wait for the 1.4.0 release.

Dweebur2, 2023-08-07:
Hello silenec.. yes I was wondering the same thing with issue #114.  Sounds the most similar to my problem.  Good advice..will wait until the latest firmware is available and try it again.  Appreciate your feedback

closed 2023-08-22 by Dweebur2