danielnilsson9/bbs-fw #56: Sport mode sometimes switches to standard mode without command

CitizenDev, 2022-09-09:
I have noticed 2 occasions now where firmware has dropped from sport to standard without me giving required input which is PAS 0 + Light button.

Light sensor has been turned off on 850C display unit.

danielnilsson9, 2022-09-10:
Can pretty much only happen if there was a reset/crash... Will likely be hard to find

CitizenDev, 2022-09-10:
I'm not sure how much additional info I can currently provide because there isn't anything indicating the change except the PAS & throttle changes so I probably did not notice it instantly because it happened at lower speeds.

My standard has throttle turned off and speed limited to 25kmh. So then I was going less or around 25 with PAS only there was nothing to indicate that it happened. Both times I noticed when throttle no longer worked.

Btw, just a bit unrelated but can it be configured so that after system restart / error last used profile would be used? Currently it will always revert to standard. I would much prefer it to stay in sport until I change it myself. And If I changed to standard, it also would remain active until change input is given.

danielnilsson9, 2022-09-23:
Any additional crashes? Let me know if you have been able to see any connection to a specific use case.

CitizenDev, 2022-09-24:
Sorry I haven't driven my bike recently because of crap weather. I'm not sure how much I will be driving at winter. But i'll definitely keep u posted when I get to drive.

danielnilsson9, 2023-07-05:
This is likely related to #114 and the same solution should apply.
A communication error with the display could cause this behavior since fallback was to return to standard mode if an invalid mode switch command was received, checksum of received command was not verified either.

danielnilsson9, 2023-08-01:
Most likely fixed in #153

closed 2023-08-01 by danielnilsson9