danielnilsson9/bbs-fw #57: Display showing 00h

GreyXor, 2022-09-10:
Hi,
i got my new BBSHD v1.5 and i just install this firmware 1.1.0.0.
is working well but my display (the default one from bafang) is not working.
i just get the message "00H" blinks.
If I connect in usb I have no error log.
Some idea ?

danielnilsson9, 2022-09-10:
Not really, never heard of error code 00, it's not an error, it is what is reported when everything is ok and bike is stationary. No idea why that would be shown on the display. Which display is it?

danielnilsson9, 2022-09-10:
Disregard my comment above, I have made a simple error which obviously haven't been discovered since no other display has reported it as an error. I will fix it and release a patch. Status code normal operation is 01 not 00...

If you can, please try the attached firmware file for bbshd to verify your error is resolved.
[bss-fw-1.1.1b.zip](https://github.com/danielnilsson9/bbs-fw/files/9540314/bss-fw-1.1.1b.zip)

GreyXor, 2022-09-10:
Now display is showing 30h and i can't drive the bike anymore. Nothing happens
and also i cannot flash back to 1.1.0.
When i click "Download" button it receives the MCU only when I turn off the battery. if I turn on nothing happens, if I turn off then it receives the MCU but then it fails because the battery has gone out. any idea what to do?

GreyXor, 2022-09-11:
No more 00h with 1.1.1b.
FYI now i get after temperature trigger this log `Unknown (5860) in log`

danielnilsson9, 2022-09-11:
Thanks for reporting, I made a new release to resolve this issue.
Yes, that normal, you will need to update you config tool to read that log.

closed 2022-09-11 by danielnilsson9