danielnilsson9/bbs-fw #60: Voltage reading calibration

danielnilsson9, 2022-09-11:
Voltage reading components on PCB is of bad tolerances causing a quite significant error on some controllers.
Add possibility to calibrate from the Config Tool.

The voltage reading is used for both LVC limiting and battery percentage calculation.

GreyXor, 2022-09-18:
My battery is at 100% rightnow, for set up the max voltage, should i trust my BMS (there is 2 volt difference between) or the value returned from controller ?

danielnilsson9, 2022-09-19:
Since no calibration has been implemented yet it is recommended to set you max battery voltage to what the controller returns at full battery charge and the low voltage cutoff to what the controller returns when you are at your desired low voltage limit.

closed 2022-09-25 by danielnilsson9