danielnilsson9/bbs-fw #61: Motor.c variable definition lvc_volt_x10 is uint8_t not uint16_t

dav0000000, 2022-09-11:
Could this be a problem with lvc?

motor.c definition
static uint8_t lvc_volt_x10;

motor.c value assignment with 42 would mean 420 and not fit in uint8_t
lvc_volt_x10 = (uint16_t)lvc_V * 10;

Then used in motor.c function
uint16_t motor_get_battery_lvc_x10()
{
	return lvc_volt_x10;
}

And motor_get_battery_lvc_x10() is called by function apply_low_voltage_limit() in app.c

danielnilsson9, 2022-09-11:
Damn yes, well no, not a problem with lvc since that calculation is done in configure:
`send_request(OPCODE_LVC, ((uint16_t)lvc_V * ADC_STEPS_PER_VOLT_X10) / 10u);`

Which doesn't use that variable. However that value is saved for use in battery percentage calculation... And that is why it has seemed a bit of when I was testing on my bike... But worked perfectly on the bench (being set at 24V) since I have a 30V power supply.

dav0000000, 2022-09-11:
motor_get_battery_lvc_x10() is called on line 603 and 607 in app.c so it that would have been getting uint8_t data too.

danielnilsson9, 2022-09-11:
Yep, I verified, it was not working at all... I patched the just released 1.1.1 version with this fix too.

closed 2022-09-12 by dav0000000