danielnilsson9/bbs-fw #63: Ebrake issue with throttle cutout error? doesnt happen with stock firmware....Bafang bbso2b

Zensaition, 2022-09-20:
Ebrake issue with throttle cutout error? doesn't happen with stock firmware....also when it cuts out the display which I have a dpc11, sky rockets the speedometer to like 90+ and goes crazy wacky out of spec... when I'm going way below that number ion speed mph or kph.

Love the throttle response and quickness, seems really dialed there, just hoping for no over heating issues lol,  just want to have the information be correct and its gonna be perfect <3

as for ideas or things to add for the firmware for the bbso2b, maybe instead of cutting throttle it can just stop the motor? would be a better alternative, no clue how bafang does it but seems to be the same or they have a bypass when it knows the ebrake is on.

Could you maybe add a firmware updater for displays to be updated as well, add more features or really tidy the ui and make it more sporty and provide more information like motor temp, range of battery, ect... If that is possible maybe Luna cycles might know as them about it?

well either way I'm gonna run this frimware for another day ,see if I can just ignore the issues but thanks so much for the work and making this platform better appreciate you, I love ebikes!!!! :D

dav0000000, 2022-09-20:
I'm pretty sure the skyrocket to high speeds on the display is the result of the firmware crashing due to an error. It only started happening to me after after one of the recent changes. Others have reported similar issues. I think it is something to do with the timer changes as I reverted back to using two timers (can only do this on the BBSHD apparently) and I have not had a single crash in well over 300km.

When the system crashes it will revert back to STANDARD mode. If you haven't changed the config then you will lose throttle until you change back to SPORT mode.

When the ebrakes are applied it bypasses the controller so cannot be controlled by the firmware. The ebrakes will always stop the motor, so the throttle will also not do anything even though the firmware can see that you still have throttle applied. If that makes sense. Daniel is the expert and will no doubt explain it better than me.

danielnilsson9, 2022-09-20:
I'm gonna need some more information here because it is not clear what you are referring to.

- Do you not expect motor power to be cut when applying brakes and using throttle?
- What exactly is a cutout error?
- Does the speed sensor go crazy every time you apply the ebrakes?

You can revert to original firmware if you have access to that, just make sure you set frequency to 18.432MHz when you flash the original firmware.

Firmware for displays can probably be done but I won't be doing any work on that as I'm not really using a display myself and have no interest in such information. But anyone is free to work on that if they want and if there is any progress on it then I would be happy to add messages for more status information retrieval.

@dav0000000 Was this issue not resolved for you, did you still have the same issue after we found the interrupt stuff? I have also 300km+ on my bike after this change and haven't seen any issue :( There are only report about the speed sensor stuff and the reset on the BBS02 as far as I know.

dav0000000, 2022-09-20:
@danielnilsson9 that is good to hear. I quickly tried the pragma version but I failed. It was late and I already had the working version with just the code for the timers removed (created before the pragma change). I suspected I had done something wrong when testing the new version with pragma. I had updated to SDCC 4.2.0 since the previous test and honestly can't recall if I wrongly set it to 18.432. Then I didn't get back to it. I will make an effort to try it again.

danielnilsson9, 2022-09-21:
> I'm referring to the display communicating to the firmware of the bbso2b from your firmware..... yes I know it cuts the motor but the original firmware had some type of way of knowing the throttle was cut off, then getting an error "Fault in throttle error 05"....
> 
> Yes the speed sensor or something in the communication gets outta wack and it shoots up to like 90+ mph on my dpc 18 display...when I hit my ebrake especially the back brakes.

I'm still not sure I understand what you are saying, can you confirm that the following is what you are experiencing:

1. You apply your ebrakes while riding.
2. After applying your ebrakes speed is then indicated wrong on the display
3. Then you see throttle error 05 on the display?

Is this correct?

> well if you are open sourcing this, I think there should be a default revert hex file to original firmware. especially if there is compatibility issues or whatever is going on till its found and fixed like in my case. I don't wanna mess up my miles or logs on my display. Or something like that.... if we need to find it a lot of people gonna have to find it hard core on some single hidden page.

I do not have the legal right to redistribute Bafangs original firmware. This firmware is provided free of charge, you are expected to read the documentation of this firmware to know about current issues and especially read though the page https://github.com/danielnilsson9/bbs-fw/wiki/Flashing-the-Firmware before flashing anything on your controller, where it is mentioned that it is a good idea to have access to the original firmware.

dav0000000, 2022-09-21:
To me a 99 speed showing on the display is due to the firmware crashing and restarting. For safety I would think the system needs to confirm a working throttle at startup. I suspect this may be actually handled by the NEC controller as I know if I turn my bike on while pedalling it will not work, I suspect the system thinks the pedal sensor is faulty so maybe it's the same with the throttle. What happens when you turn the bike on with the throttle on? Does it produce that same throttle error. Of course if you test this then lift your rear wheel off the ground.

@Zensaition the timer changes I mentioned were to use the same timer for sensors and system timer. There is only one timer for this on the BBS02b. I have the BBSHD.

One other thing, do you have a gear shift sensor?

dav0000000, 2022-09-21:
@danielnilsson9 I wonder if sensors_init SET_PIN_QUASI(PIN_SHIFT_SENSOR); could be related if it is not the timers. Reverting the timers worked for me but could it actually be this setting causing the issues. The gear sensor worked fine on SET_PIN_INPUT. I suspect you found a reason to change the brake to quasi and just changed the shoft sensor at the same time, I would be interested to know if you changed these two pins from input to quasi for a specific reason.

dav0000000, 2022-09-21:
@danielnilsson9 everything is working great for me. The shift sensor is working for me but I don't use the timer changes or any of the changes to compensate for the BBS02. I do have QUASI on brake, uart and shift sensor pins. I have not tested v1.1.1 again. I thought the shift sensor could be somehow related to this problem as I thought you don't have one. I may not be your smartest user but I have been running your firmware for well over a year and not gone back to bafang software even once. I love your work and don't want to be confusing.

danielnilsson9, 2022-09-21:
@dav0000000 Alright thank for confirming, sounded like you had tried it and still had issues. Let me know how it works for you if you ever decide to update (I know, it's a pain to disconnect display to flash...). I do appreciate the testing you have done, you have been very helpful in finding multiple issues.

Given that he says the issue often occurs when the back brake is applied I have the following idea:

Back brake (assuming disc brake) will shift the wheel slightly when applied. Maybe the speed sensor is then a bit out of range causing an unstable speed sensor reading (short pulse) triggering some error in the speed calculation causing a reset? Sounds a bit like a stretch but I will see if I can find anything.

dav0000000, 2022-09-21:
@danielnilsson9 I just tested the latest v1.1.1 and did not see any issues this time. It was only a 5km ride but looks like it's all sorted now. Thanks again.

Zensaition, 2022-09-22:
@danielnilsson9  @dav0000000 Gotcha sounds good to me guys I'm glad your figuring it out lol. yes I do have a shift sensor....I'm just going off what I'm seeing and feeling from the firmware, no clue the technical no how. but I'm hopeful for 1.1.2 lol.

I'm wondering if maybe its my settings as well but your probably right that its a conflicting issue with the speed sensor and throttle spazzing out

@dav0000000 could you give me a example or picture of what I should use for the timer like a number?

danielnilsson9, 2022-09-22:
That is... interesting? Makes no sense, looks almost like your e-brakes are connected to the throttle signal and not do the brake signal on the controller.

I can reproduce you log by shorting the brake signal with the throttle signal but that also causes my motor to spin when throttle is at zero, which doesn't happen on your bike I assume?

You either have a very wired installation, or bafang did something different in earlier versions or you have a short circuit between you throttle and brake signal (which doesn't make a lot of sense if you are not seeing power applied when not touching the throttle).

This is my log when applying e-brake on my BBS02B controller:
[image]

The motor target current is expected to go to 0 as a safety precaution, motor controller report brakes applied (the red error message, which is not really an error).

And I fail to see how this can be related to the speed reporting going crazy, it is definitely not the crash you got that I was expecting... might be two different issues you are encountering.

It is really a BBS02B controller you got? Any pictures of your bike and installation might be helpful if you are willing to share.

Zensaition, 2022-09-22:
@danielnilsson9 yeah its weird that yours is different then mine, I'm sure I installed your firmware so should do the same... yeah its a 2017 bbs02b controller and core so idk? its from luna cycle its top wattage was at 1500w so its the upgraded one I'm sure??.

yeah idk I always thought this was normal coming from it as I think the stock firmware did the same thing? didn't know there was a brake signal then just cutting the throttle like I thought bafang did..... it could be because of my 2 pin Magura mt2 ebrake? is there suppose to be 3pins for ebrake? maybe 2 pins cut throttle instead?

as for pictures I just woke up... but what do you need?

Zensaition, 2022-09-22:
@danielnilsson9 Okay I did some research its my 2 pin ebrake is different, as its not a hall effect sensor but just a switch for motor on and off so good to know , the third pin is for brake signal :D

"[Re: Brake Lever Signal Difference 2 vs 3 pin] (https://endless-sphere.com/forums/viewtopic.php?t=113004#p1670950)
by [HK12K](https://endless-sphere.com/forums/memberlist.php?"mode=viewprofile&u=66524&sid=2c13da73fcba8e5a74ce852ba63fddc2) » Aug 14 2021 12:47pm

"The 2 wire one is a momentary switch that acts like an on/off button. All or nothing. The 3 wire one uses a hall sensor and is for variable regen braking. They are not interchangeable. If your controller does not have an input for a 3 wire you cannot use a 3 wire lever. Two different animals completely.

That said if you are going to re-use your existing 2 wire switch you can modify the cable by soldering on an appropriate connector to either end, however it will only be making use of 2 out of the 3 wires."

[image]

danielnilsson9, 2022-09-23:
That makes a lot of sense!  didn't even know that type of harness existed. Most likely is the switch connected inside the harness to cut the throttle signal instead of activating the brake signal.

I don't think the low throttle error is implemented in the Bafang original firmware and therefore your are not seeing an error there. I will provide you with a custom firmware build later today to disable that error and try to fix the speed sensor issue, it would be helpful if you were willing to give that a test.

danielnilsson9, 2022-09-23:
Please try attached firmware and run it for a bit and reports how it works for you.

- Low throttle signal error disabled.
- Low throttle signal is used to indicate brakes active.
- Better check for invalid speed sensor signals to see if it helps with your speed display issue.
- Throttle low signal used as a brake signal, e-brakes should work for PAS too as long as you have a throttle connected.

[bbs-fw-1.1.1p.zip](https://github.com/danielnilsson9/bbs-fw/files/9635514/bbs-fw-1.1.1p.zip)

Zensaition, 2022-09-24:
@danielnilsson9 sry been busy but I'm back and glad it helped you as much as it helped me learn more as well lol. 

I will install rn and see what happens, thanks so much for you looking into this, helping me and others with this type of brake setup and the speed issue. ill write back what I find out out about the build firmware. thanks again XD

Zensaition, 2022-09-24:
@danielnilsson9 So I updated the firmware to the custom one you provided, the error is fully gone now but the speed is staying on screen and not going to 0 right away it takes like 3s to go down to 0 when I apply the ebrakes...also max speed seems to be only 35mph from stand still, dont know why? maybe its the display preventing it to go further even though it goes to 99mph lol? but so far looking better :D . as for the speed sensor... no clue the only settings I see is "signals per rotation which is set to 1....idk dav was talking about system timer and senor timer which isnt in the config tool soo cant do much there...

so an improvement for sure. The controller is still doing some weird micro adjustments of throttle current from like 3% to 7% depending, from pressing the ebrake or a bunch? like I showed earlier. but close to perfect really just needs some few tweaks and its golden XD. ill take it for a spin to see if I find any thing later today, around noon pacific time.

Thanks a bunch

danielnilsson9, 2022-09-24:
The speed shown on display is rather slow to update since the display is doing some averaging to avoid showing jumpy values, that stuff entirely done in the display firmware and isn't really affected by the controller firmware, I believe you would have the same thing with original firmware. It's more noticeable when you are not riding the bike, maybe you haven't thought about it before.

Not sure about the max speed, should be set to 60mph in the default config, what is configured in your display is not used by this firmware. Could you get it above 35 mph before when the back wheel was of the ground? I guess it depends on your gearing, what is your wheel size?

I guess the BBS02 has been sold with that 2pin harness somewhere at some point but I cannot find it for sale with that harness at the moment. The 3pin ebrake seems to be the standard way now which triggers the brake signal into the controller instead of cutting the throttle signal.

Zensaition, 2022-09-24:
@danielnilsson9  gotcha that makes sense just, it would cut the motor and the display would go to 0 immediately before but I get you have limits so I wont press on the speed unless there is a way to make it slightly better or a lower timer of the speed adjustments of some sort?

yes when the back wheel is lifted of the ground max is 35mph, the config is set to 100kph....if its in 60mph, it goes to 17mph on the display lol....idk about that something is going on here..... my wheel size is 700c 28, gearing is on 5

oh no the harness is 3pin always, I believe its just the brakes connecting can be 2 pin or 3pin they just plug into those 2 configs of the brake signal, 5v or the 5v with ground, sry for that confusion.

but yeah thats really all I see from a ground stand point havent gone and road tested yet sooo but hope this helps you in some way

danielnilsson9, 2022-09-24:
Sound like I broke something then... I guess you have this correct but your display is set to show imperial units and you have also set imperial units in my configuration tool right? You must set the same in display and configuration tool or the speed shown will be wrong. This firmware is metric units by default so you must change it if you have your display set to imperial units.

Okay, so you have modified your ebrake plugs to be connected between the black and white pin on the ebrake cable from the harness as shown in the thread you linked?

Because from your log you do not have a brake signal triggered at all,  it just shorting the throttle signal to ground which makes no sense if you say you have it connected to the brake signal?

Zensaition, 2022-09-24:
> 
I think the config tool converting to imperial is broken on the bbso2 side of it, that's why also inputting it to metric and the display is already doing that conversion is fine. but fixing it would be good on the config tool.

as for my ebrake they are put in the motor cut off configuration so the 5v and ground lead.

no as I'm not using the brake signal lead, I would have to resolder/wire the brakes for that. if you cant fix it for the this config, ill do it. but just saying other people might have this so might be good to do it.

I never said it was connected to the brake signal I needed to do so for the high level brake or regen as the post said....then the motor cut off config I currently have...

as for the connection I read its the 5v wire with the brake signal lead to get the highbrake....not ground. ground is to short the throttle to motor I believe.

Well these bugs are effecting each other and maybe its good, I had this issue with the motor cut off config, to help further the firmware lol. in my last post it would be nice if you implemented the settings and things I suggested as did Dav saying which would help the speed sensor issue as well.

danielnilsson9, 2022-09-24:
So let me get this clear, you have opened an issue with a very vague and hard to read description of your problem related to e-brakes not working.

You have failed again and again to answer questions in a way that can be understood and now you are telling me you have modified the wiring and not even connected your e-brake lever signal to the brake signal on the controller? You didn't think that was something you should have mentioned originally, especially since you were having issues with stuff you had modified from the standard setup?

I give up, this was a big waste of my time. Please feel free to fork this project and make the necessary modification you need, they will not be part of this firmware.

I hereby close this issue.

closed 2022-09-24 by danielnilsson9

Zensaition, 2022-09-24:
this is still an issue lol because the stock firmware didn't have the issue also my brakes are stock I have to mod and rewire them to transfer over to the other config.....you are going to conclusions Jezz man chillout wtf....if you have been reading....

Zensaition, 2022-09-24:
This is still an issue lol there are 2 pin bafang brakes as well are you daft..... because the stock firmware didn't have the issue also my brakes are stock I have to mod and rewire them to transfer over to the other config with the brake.....You are going to conclusions, man chillout wtf.... Well at least fix the display throttle to ebrake throttle current issue....even the metric conversions... I get its a lot of effort just take a break and breathe. don't blow up at me when I'm just being honest.,