danielnilsson9/bbs-fw #64: Electric scooter style Cruise Control

silenec, 2022-09-25:
Would it be possible to have this kind of cruise control as in the video? It works like this on most electric scooters and it is really simple and intuitive.

https://www.youtube.com/watch?v=W3KI9TGR7zo

You basically hold the throttle at some (0-100%) position for a few seconds and cruise control activates / holds this throttle level or in most cases it holds the achieved speed for you.

To deactivate, touch throttle again, pedal backwards or brake (if you have brake sensors installed).

This would only work in the Cruise "Assist Level Type" or you could even add new all-in-one Assist Level Type called "PAS & Throttle & Cruise".

Or even "PAS & Cruise" and "Throttle & Cruise" if someone wanted that for some reason I guess.

There is probably no buzzer in the display like on electric scooters so there would not be any audible confirmation but maybe something could be at least displayed on the LCD during activation? Like current speed changed to "99" for a few seconds would help as a visual confirmation when cruise activates or maybe "haptic feedback" from the motor could be implemented - once cruise activates, set motor power to 1/4 of the current output for a split second and then go back to selected cruise power. That would be nice.

danielnilsson9, 2022-09-25:
Sounds like a much better implementation! The existing one is not very useful, I'm not even using it myself...

I think it would be best to regulate the power on the selected speed when cruise is activated, and not just hold a static power level.

silenec, 2022-09-25:
Yes, most, maybe all scooters with this function try to stay at the speed selected when cruise is activated so that might be the best way to go. I would however still like to be able to set maximum Target Power and Speed for different PAS/cruise assist levels.

On my scooter I have 3 configurable power / "PAS" levels with different max power and max speed and I can even switch between them while cruise control is active. So for example first goes max 10, second 25 a the third one 30km/h and I can control my speed when I want to go faster / slower with just a press of a button without even touching the throttle, without reactivating cruise. It is kinda fun :)

manusalgam, 2023-04-19:
Why not enable it by braking->release brake WHILE stands on throttle in a fixed position (max deviation=5-10%, power/speed value taken by average or last) during X seconds (3-5 it's fine)?

Throttle may allow to grow up speed without disabling cruise (as some cars, useful for overtaking in motorways and keep last speed without enabling it again), and disable it just braking or pedal backwarding

danielnilsson9, 2023-04-27:
The suggestions here are a nice improvement but I will not spend the time to implement this, if anyone want to give it a go, feel free to do so.