danielnilsson9/bbs-fw #66: FW 1.2.0 - PAS does not kick in until going 6mph

cdbrow, 2022-10-07:
I just recently upgraded to 1.2.0 on my new BBSHD install. For me the PAS doesn't kick in until I'm going 6mph. I would like it to kick in immediately like the original FW did.

danielnilsson9, 2022-10-08:
There is now lower speed limit for when PAS starts. However the PAS stop delay setting will affect how fast you have to pedal in order for pedal activity to be detected, since there is only 24 pulses on a full rotation. This means you cannot set the PAS stop delay too low or pedal activity will not be detected.

The default settings of 200ms will require you to have a cadence of 12.5 rpm.
60 / (24 * 0.2) = 12.5rpm

If you set it to 50ms you will be required to have a cadence of 50rpm.

60 / (24 * 0.05)) = 50rpm

Have you adjusted the default value of 200ms? If not and it is still to low, try setting it a bit higher.

cdbrow, 2022-10-08:
Thanks for getting back to me, I did have it set to 50. I think I may not understand the setting correctly. I am wanting the motor to cut off as soon as possible when I stop peddling. I also want it to start as quickly as possible when I start peddling. What are the best settings to accomplish that? Thank you! I did donate. :)

danielnilsson9, 2022-10-08:
I understand, the short answer is that it is simply not possible with the sensor that is present in the motor. For every revolution the sensor will send 24 pulses. The only way to detect if you are no longer pedaling is by the absence of a pulse, i.e. the pedals have not been rotated 1/24 of a full revolution in x amount of time, where x is the value of the setting "Stop Delay".

But when you set it too low then you run into the issue of where pedaling slowly triggers the PAS to stop since there has not been any pulse in x milliseconds.

If you plan backwards slightly it will stop instantly however, since it possible to detect that with this sensor.

cdbrow, 2022-10-08:
From your experience, are the defaults for start and stop the absolute lowest to achieve what I’m looking for?

cdbrow, 2022-10-11:
I'm happy with it now, I have the start delay set to "0" and the stop delay set to "200". Your explanation makes sense to me.

closed 2022-10-11 by cdbrow