danielnilsson9/bbs-fw #68: I'm sorry I don't know what's going on with you controller. You can feel the motor power increase and decrease just when using PAS and no throttle? Sounds like some form of limiting is going on, please check that you temperature sensor reports normal values in the event log in config tool.

rajko68, 2022-10-24:
I'm sorry I don't know what's going on with you controller. You can feel the motor power increase and decrease just when using PAS and no throttle? Sounds like some form of limiting is going on, please check that you temperature sensor reports normal values in the event log in config tool.

_Originally posted by @danielnilsson9 in https://github.com/danielnilsson9/bbs-fw/issues/55#issuecomment-1242274336_

rajko68, 2022-10-24:
Hello Rajko from Slovenia. I installed version 1.2.0. However, there is a problem with the throttle. the engine starts to whine and finally the display turns off

rajko68, 2022-10-28:
bbshd ,display DPC18,batery 52 volt.How to run
the auto adjust app.I set 58v in config.where to set current.

dav0000000, 2022-10-28:
Max Current is set in the APP. If I set this to 33amps I can trigger a BMS error from my 30amp max battery if I go full power for too long. Then my display will turn off. Did you change max current from default? If you did then you need to be sure your battery supports the extra current. It's worth checking the max current your battery is rated for to be sure. The Wiki explains where to change max current so best check that if you haven't already.

rajko68, 2022-10-29:
can you send me setup suggestions

rajko68, 2022-11-02:
I lover the current to 28 A .But when batery goes to 50 % is the same .Display cut off.

danielnilsson9, 2022-11-02:
Ok, I cannot help you. The controller cannot cut power to the display as the display is directly powered from the battery so unsure what is happening. You should probably revert to the original firmware if that was working for you before.

rajko68, 2022-11-03:
It was the same whith original firmware.Becose this i put your.

danielnilsson9, 2022-11-03:
That would be information that would be a good idea to disclose when you open an issue like this. Sounds like your battery is broken to me, voltage sag, broken cell or something triggering low voltage protection in the battery BMS.

closed 2022-11-04 by danielnilsson9