danielnilsson9/bbs-fw #70: Bbs02b tuned to max 3000w and 50amp controller

Bafanger, 2022-10-28:
Once i was using the original software i was getting 3000w peak. Now im getting max 1450w. The motor jas plenty of torque but i barely go over 30mph

dav0000000, 2022-11-01:
3000 watts sounds expensive. Cool story though. One time I saw a fox.

danielnilsson9, 2022-11-01:
Sounds like a bad ideá, but post boot logs from the config tool and maybe I can help you.
Also post you configuration.

See last section here:
https://github.com/danielnilsson9/bbs-fw/wiki/Configuration-Tool#event-log

Zensaition, 2022-11-22:
> Once i was using the original software i was getting 3000w peak. Now im getting max 1450w. The motor jas plenty of torque but i barely go over 30mph

Bro post your stuff lol

danielnilsson9, 2022-12-04:
No update, closing.

closed 2022-12-04 by danielnilsson9