danielnilsson9/bbs-fw #73: Req: Android config tool

rayholland, 2022-11-13:
Would love to see an android config tool.

Bluetooth communication would be possible with this cheap Ble Bluetooth to Uart Transceiver Module :
https://www.go2tech.nl/winkel/componenten/at-09-bluetooth-4-0-ble-module-cc2540-cc2541/

Cheers,
Ray

danielnilsson9, 2022-11-14:
Would be nice yes, I don't have time to work on an android app. But if anyone wants to do it I'm available for questions, protocol is not docummented but the messages can be found in extcom.c.

As for Bluetooth, you will probably have to power off or disconnect you Bluetooth module while display is connected or it will interfere with the controller -> display communication.

I solved that by connecting Bluetooth module on a UART proxy, code for the proxy is in src/logger but it is not really maintained atm. I only used the Bluetooth for viewing event logs on my phone while riding.  An Arduino was used for the proxy implementation. The best implementation for a proxy like that would probably be to use an ESP32 which has both Bluetooth and 3 UARTS (which is needed) in the same chip.

Everything was powered of the 5V line from the controller which worked well, there seems to be enough current.

jvz1990, 2023-11-14:
Hi @danielnilsson9,

I'm kinda keen on this idea. I have a few cc25xx dev boards lying around. Or an ESP type device is more user friendly? 

Q. Could you explain what the PL line is for and how it works?

Conceptionally the hw solution would involve a splitter between the BLE device and with a switch between the P+ lanes.

SW would indeed be time consuming to develop another firmware for the BLE device and Android app.

danielnilsson9, 2023-11-16:
@jvz1990 

PL sends power to the controller. P+ is connected directly between battery and display. When display is powered on through a button it pulls a transistor which connects P+ to PL and then controller also powers on.

I had a private project like that before I wrote this firmware which replaced it. I didn't really have any use for the android app, only used it to unlock unrestricted limits. The controller was reconfigured in the same way as EggRider does it nowadays.

It used a ESP32, I will attach the drawings for my pcb to give you an idea what would probably be needed.

* You need a power supply circuit for 48V -> 5V/3.3V
* Level converters for 5V -> 3.3V (maybe not really needed, depends on input tolerances on ESP/cc25xx) 
* Transistor to power on controller P+ -> PL
* OPAMP in schematic is for battery voltage measurement, not needed

On the other hand, if you can accept that your controller is always on when battery switch is on you might get away with using 5V line from controller to power external microcontroller/bluetooth. Then everything is much simpler. But I'm not sure how much power the 5V line can deliver, the 6V light output is also an option though.

[image]
[board.pdf](https://github.com/danielnilsson9/bbs-fw/files/13382475/board.pdf)
[schematic.pdf](https://github.com/danielnilsson9/bbs-fw/files/13382476/schematic.pdf)

My own interest for doing something like this together with this firmware is low but I will help with implementation of necessary data export and new commands for the controller firmware if someone decides to work on this.

Damien0505, 2023-12-17:
Someone looks like they are already planning a new mobile app.  Sounds good if/when it comes to fruition!
https://github.com/S-CODE-pl/ChainBreaker