danielnilsson9/bbs-fw #79: Allow 100% cadence during PAS throttle override

danielnilsson9, 2022-12-21:
Common use case which is not supported at the moment.
New assist level parameter or enforce behavior if throttle current set to 100% on a PAS assist level?

EigenMania, 2022-12-22:
Slightly different idea, but related. What if "Max Cadence (%)" was split into "Max Throttle Cadence (%)" and "Max Pedal Cadence (%)". Then when you have either throttle or throttle+PAS, rpm limiting + ramping is done based on "Max Throttle Cadence (%)", but when just pedaling it's based on "Max Pedal Cadence (%)". 

For example, I keep "Max Cadence" around the 40-50% region because I like to pedal with the bike instead of having it out-spin me. But I've always felt it would be nice if the throttle could bring me just a bit slightly past my desired rpm (for quicker starts when I am lazy). Having it as an assist level parameter would be nice, so that one could have the max throttle RPM relative to the pedal RPM (my desired use case).

I think there could additionally be a global throttle current over-ride that sets a higher cadence limit only when the throttle is maxed out. Then I'd do this with one boolean (to enable / disable), and then one uint8 where users could choose the cadence % over-ride (from 0-100).

Is it silly / too much to have both?

EigenMania, 2022-12-25:
So in the end I added two config parameters to each assistance level: "Throttle Cadence (%)" and "Max Throttle Cadence (%)". The first over-rides the motor cadence limit when throttle is applied, and the second overrides only when the throttle is fully maxed out. Both can be disabled by setting their values to the PAS cadence limit at any assist level. 

Overall it's a super nice addition. Now since the throttle cadence limit can be higher than the PAS cadence limit, you can pedal along at a reasonable cadence, but the throttle keeps lots of pep ;) Really good suggestion :D 

Here's the branch: https://github.com/EigenMania/bbs-fw/tree/feature/max_throttle_cadence

danielnilsson9, 2022-12-29:
I was trying to avoid adding any more parameters to the assist level configuration as the total configuration size grows by at least 20bytes per added parameter. I can therefore not motivate having two parameters per assist level for this feature since I find it quite insignificant. But I want to support the common use case of allowing full cadence during throttle override as it is supported in the original firmware.

The assist level config has been reworked in the TSDZ2 branch in order to incorporate torque sensor support. And the assist level flags are now used more as flags than an enum. I will probably add a flags "ThrottleMaxCadence" to the flags and a checkbox to the assist level configuration in config tool to allow the user to select max throttle cadence override.

EigenMania, 2023-01-04:
The feature can be disabled if "ThrottleMaxCadence" is just set to match "Max Cadence" in the config. Maybe it's not intuitive, but it would save adding 20 bytes just for booleans, and clear instructions could be given.

danielnilsson9, 2023-01-04:
I will have to think about this a bit more, I'm damaged, we have 500k+ lines of configuration files at work 😃

danielnilsson9, 2023-01-04:
I checked you branch feature/max_throttle_cadence... You have 1 byte of xram left and maybe a few bytes of iram. Heads up, you will probably not be able to build that if you rebase from master since I added a parameter for torque amplification to each assist level.

I guess something has to be done about memory use at some point, i will create a separate issue.

EigenMania, 2023-01-13:
Here are some ideas: 

1) Enabling the throttle can be done with the assist level flags. So the variants under PAS can be {Cadence, Cadence+Throttle, Torque, Torque+Throttle, Variable}. This would save 20 bytes in the config that are used to enable the throttle (a bit of a waste IMO to use a full byte for a boolean value). 

2) 100% cadence at max throttle over-ride could be configured on the System tab for Standard and Sport assistance and apply to all assistance levels within those sets (thus only using 2 bytes for booleans).

3) If this feature should be configurable for each level, then instead of the boolean to enable throttle (from point #1), we would add a parameter for "Cadence at Full Throttle (%)" so that the cadence limit can be increased when the throttle is maxed out, but then users could optionally modify it to lower cadences if inclined. 

Point #3 is probably over-kill, but the combination of the first two points could make the feature fit without further memory optimizations (though I agree memory should still be addressed at some point).

danielnilsson9, 2023-02-01:
I will close this for now since the basic function is implemented. I will consider reopening this in the future if there is more people who feels more configuration options could be useful.

closed 2023-02-01 by danielnilsson9