danielnilsson9/bbs-fw #83: Power meter does not show anything above 10xx Watts + Ocassional crashes

Gonespa, 2023-01-02:
Hi,

I have recently installed the 1.2.0 version on my BBS02B unit. FYI this is the information provided prior to the update by the STC software:

MCU type: STC15F2K60S2
F/W version: 7.2.5S

Current H/W Option:
  . Current system clock source is internal IRC oscillator
  . IRC frequency: 18.443MHz
  . Oscillator gain is HIGH
  . Wakeup Timer frequency: 35.150KHz
  . Do not detect the level of P3.2 and P3.3 next download
  . Power-on reset, use the general power-on delay
  . RESET pin behaves as I/O pin
  . Reset while detect a Low-Voltage
  . Thresh voltage level of the built-in LVD : 3.82 V
  . Inhibit EEPROM operation under Low-Voltage
  . Hardware do not enable Watch-Dog-Timer
  . Watch-Dog-Timer pre-scalar : 256
  . Watch-Dog-Timer continue count in idle mode
  . Program can modify the Watch-Dog-Timer scalar
  . Do not erase user EEPROM area at next download
  . Do not control 485 at next download
  . Do not check user password next download
  . TXD is independent IO
  . TXD pin as quasi-bidirectional mode after reset
  . P2.0 output HIGH level after reset
  . Reference voltage: 1231 mV (Range: 1150~1320mV)
  . Testing time: 2020-6-16

And then, once programmed:

H/W Option upgrade to:
  . Current system clock source is internal IRC oscillator
  . IRC frequency: 20.021MHz
  . Oscillator gain is HIGH
  . Do not detect the level of P3.2 and P3.3 next download
  . Power-on reset, use the extra power-on delay
  . RESET pin behaves as I/O pin
  . Reset while detect a Low-Voltage
  . Thresh voltage level of the built-in LVD : 3.82 V
  . Inhibit EEPROM operation under Low-Voltage
  . Hardware do not enable Watch-Dog-Timer
  . Watch-Dog-Timer pre-scalar : 256
  . Watch-Dog-Timer stop count in idle mode
  . Program can modify the Watch-Dog-Timer scalar
  . Do not erase user EEPROM area at next download
  . Do not control 485 at next download
  . Do not check user password next download
  . TXD is independent IO
  . TXD pin as quasi-bidirectional mode after reset
  . P2.0 output LOW level after reset
  . Reference voltage: 1231 mV (Range: 1150~1320mV)
  . Testing time: 2020-6-16

  . MCU ID : F408C45C04D203
  . MCU type: STC15F2K60S2
  F/W version: 7.2.5S

  . Set frequency: 20.000MHz
  . Adjusted frequency: 20.021MHz
  . Trim error: 0.104%

I have the max. Amperage set to 30 Amps. When using the throttle (did not check this with PAS as it is not as easy to test) in a high load situation so that full power is requested, I noticed the screen (500C model) starts displaying weird values (like 88, 132, and so, very low values). Motor itself works fine, it stills feels like it is pushing as hard as it can so it looks like it is somehow not converting the  W value correctly? or perhaps it is not sending the correct messages to the screen. I have never been able to see anythig above 10xx Watts in the screen. My battery is 48V, 50.1 V fully charged so I was expecting to see about 1500W or slightly less with the locked wheel.

Perhaps related to this (or not) I noticed that the controller sometimes crashes on these high load situations. It stops pushing completelly for like a second or so (zero Watts and Zero Kph are shown in the screen during that situation) and then resumes working correctly (might be the watchdog doing his job?). Screen stays on in that situation (does not look like a battery low power situation, there is no BMS in my battery). I wonder if the program crashes during Watts value calculation?.

One repeteable way to test this is: 
-Lock the real wheel (My rear brake was able to do this)
-Apply full power for a few seconds:
-I can feel then the motor pulling and then, a few seconds later it looses all power for a few moments and then starts again pulling.

danielnilsson9, 2023-01-02:
Hi

I have a 500C display somewhere, I'm gonna try and see if I can reproduce that. It might be an issue with the display as the power calculation is done on the display and not in this firmware. This firmware only sends the battery current drawn in half amp unit and I can not see any reason for it to produce the wrong value in my code when looking at it.

As for you motor cutting out. You should definitely not stall you motor and push max throttle, you have a high risk of blowing your controller by doing that. To be honest, it sounds like you have a battery power issue but I don't know. You can try connecting the config tool and view the log while using throttle and see if low voltage protection is kicking in.

What type of battery do you have? 50.1V seems really low for full charge for a li-on battery? That should be above 54V.

danielnilsson9, 2023-01-02:
I did some test on my 500C and yes, it's definitely broken. It seems to start over from 0 when power gets above 1500W.
The dpc-18 does not show anything above 1500W (since the graph ends there) but it does not start over so the problem is in the 500C display firmware. Nothing we can do about that.

Gonespa, 2023-01-02:
My battery is a raw Hobbyking Lipo pack in 12S configuration for a total of 10Ah. Topped up it gives 50.4 V (not 50.1, sorry) that is 4,2 V x 12.
It is for sure able to pull 30A and there is no BMS limiting anything.
I have 41V set as low voltage limit.
In any case, as I have around a Wattmeter , I am planning on using that. It will give me maximum an minimum Voltage an current so I will be able to register the Sag under load, just in case, I will let you know.
I will try to check the log as well as I test again.
As for the display, it,s Ok, I can live with that.

Thanks.

Gonespa, 2023-01-02:
Hi,

I have just retested the scenario with the logging activated.
An error is shown exactly when I notice the cut.
Locked the wheel and accelerator set to max:

```
2023-01-02 23:05:09.226    Info    Temperature, motor=0C, controller=17C.
2023-01-02 23:05:09.491    Info    Battery voltage reading, value=46,2V.
2023-01-02 23:05:12.660    Info    Motor target current changed to 2%.
2023-01-02 23:05:13.970    Info    Motor target current changed to 9%.
2023-01-02 23:05:13.970    Info    Motor target current changed to 20%.
2023-01-02 23:05:13.970    Info    Motor target current changed to 33%.
2023-01-02 23:05:13.970    Info    Motor target current changed to 50%.
2023-01-02 23:05:13.970    Info    Motor target current changed to 75%.
2023-01-02 23:05:13.970    Info    Motor target current changed to 96%.
2023-01-02 23:05:14.287    Info    Motor target current changed to 93%.
2023-01-02 23:05:14.415    Info    Motor target current changed to 90%.
2023-01-02 23:05:14.670    Info    Motor target current changed to 86%.
2023-01-02 23:05:15.184    Info    Motor target current changed to 83%.
2023-01-02 23:05:16.083    Info    Motor target current changed to 80%.
2023-01-02 23:05:16.146    Error    Motor controller status changed to 0x20.
2023-01-02 23:05:16.731    Info    Motor target current changed to 75%.
2023-01-02 23:05:16.731    Info    Motor target current changed to 60%.
2023-01-02 23:05:16.731    Info    Motor target current changed to 45%.
2023-01-02 23:05:16.731    Info    Motor target current changed to 30%.
2023-01-02 23:05:16.731    Info    Motor target current changed to 15%.
2023-01-02 23:05:16.731    Info    Motor target current changed to 0%.
2023-01-02 23:05:19.199    Info    Temperature, motor=0C, controller=18C.
2023-01-02 23:05:19.508    Info    Battery voltage reading, value=46V.
```
It is weird that it looks like I released the accelerator at some point before the error is shown, but that is not the case. I did after that.

Low voltage limit is not triggered. (I was able to trigger and see the message in a previous test, so I know that works as expected, but in that case I did not notice the cut nor any error was shown in the log).

Any clue when does that Kind of error show?

danielnilsson9, 2023-01-02:
Interesting. Yes, according to my notes here https://github.com/danielnilsson9/bbs-fw/wiki/NEC-Motor-Control-Protocol that is a current sense error.

I suspect you are hitting some hardcoded limits set in the secondary microcontroller which runs another firmware. Maybe it is not possible to run BBS02 with 30A config then. I don't own a BBS02 (only a controller) so never tested it under load really.

Gonespa, 2023-01-02:
I see, that must be the case. I will lower the Amps to 25 as the legacy firmware allows and recheck, just to confirm this. It should not trigger any error in this case, as I remember doing this kind of test when I installed it.
It certainly looks that way.
I any case, your firmware works much better than that, much smoother, allows the use of PAS + throttle, gets rid of that absurdly slow ramp up when engaging the PAS, etc etc.
I will close this as there is really nothing to do about it.

Kind Regards.

closed 2023-01-02 by Gonespa