danielnilsson9/bbs-fw #85: This needs quotes

EigenMania, 2023-01-04:
https://github.com/danielnilsson9/bbs-fw/blob/23cf98dacf188b936fdeb7806518f22440a65a72/src/firmware/makefile#L60

danielnilsson9, 2023-01-04:
hmm weird, it works for me both in visual studio and from cmd line

EigenMania, 2023-01-04:
"make clean" on cmd and "Clean Solution" on Visual Studio both fail for me without quotes. I thought first that maybe echo is trying to execute the line "Clean Finished" since clean might be a keyword, but it also fails with random words. But then maybe it's something with my system that I haven't figured out...

[image]

danielnilsson9, 2023-01-04:
Ok no idea why that is happening for you but I will fix, not sure why I removed the quotes to be honest

danielnilsson9, 2023-01-04:
Which make version do you have?
`make --version`

Because if I add the quotes it starts printing the quotes too which was not intended.

EigenMania, 2023-01-04:
GNU Make 3.81, built for i386-pc-mingw32

EigenMania, 2023-01-04:
Oh shit that's from 2006...

danielnilsson9, 2023-01-04:
Strange, same as me, but I found a fix and removed echo and used the makefile print statement instead

EigenMania, 2023-01-04:
Great, that worked for me too

closed 2023-01-04 by EigenMania