danielnilsson9/bbs-fw #86: Optimize memory use

danielnilsson9, 2023-01-04:
There is basically no ram left when building for BBS02 (which only has 1792 bytes of xram compared to 3840 bytes on BBSHD).

Things that should be fixed:
* msgbuf in extcom.c , uses 192 bytes of ram at the moment when uart rx is already buffered with 256 bytes.
* rx/tx buffers in uart.c. Uses 256 bytes each and there are 4 of them contributing 1024 bytes of ram usage.
(buffers are this size since the math was atomic avoiding issues with isr:s if I remember correctly)
* MAP16/MAP32, make into functions, should decrease ram usage since static allocation is used when compiling (even for function scoped variables)

EigenMania, 2023-01-24:
Regarding the config size: 
- All of the booleans in the System tab (currently 5) could be combined into 1 byte, saving 4 bytes and leaving space for 3 more. 
- Add assist level Type "PAS+Throttle" instead of "Enable Throttle" would save 20 bytes 

If you would be in favor of these adjustments, I can implement a PR.

danielnilsson9, 2023-01-24:
1. Could be done but feels a bit silly since there is > 700bytes that can quite easily be freed up by changing a few things in the uart drivers.
2. Not sure what you mean, I think it is already implemented. Enable Throttle checkbox sets the ASSIST_FLAG_THROTTLE in the assist level flags variable.

So I would say that the goal for this issue is to fix the uart drivers, then there will probably be enough free ram for a long time. All other code except the uart driver is currently using a totol of ~700 bytes of ram.

danielnilsson9, 2023-05-30:
896 bytes of memory freed up by reducing uart buffer sizes, closing this.

closed 2023-05-30 by danielnilsson9