danielnilsson9/bbs-fw #92: Throttle Speed Control

EigenMania, 2023-02-02:
Original firmware allowed for a speed control mode with the throttle input. This could be a boolean option on the system tab, and throttle could be mapped to target_cadence accordingly.

danielnilsson9, 2023-02-12:
It was intentionally skipped since I found it unusable. I'm not sure anyone actually used that, did you?

EigenMania, 2023-02-12:
I find it quite convenient for low-speed riding (stop-and-go traffic, parking areas, and so on...) because you can modulate the speed directly with the throttle, whereas with PAS or torque throttle, the bike will accelerate to the maximum set cadence even at low current levels. I don't know exactly how / what the original firmware did for speed mode. I experimented by just setting the target cadence proportional to the throttle input, and this worked nicely.

closed 2023-04-13 by EigenMania