danielnilsson9/bbs-fw #97: Eggrider V2 loses connection when entering programming mode.

Jamberou, 2023-03-04:
Hello, has this firmware been tested with eggrider V2? For me it had the problem described in the headline.

danielnilsson9, 2023-03-08:
Hi

Yes this firmware does not support programming changes through egg rider at the moment. Some form of support is planned but it is not yet clear to what extent.

Perlebikes, 2023-03-10:
Hi, 

Is it possible to switch the mode between offroad and road using the egg rider button? 

Your firmware also offers two different modes, right? 

THX! 

BRGDS
Christoph

Jamberou, 2023-03-16:
Hey, sorry i did not test the button and i already changed the firmware back to original. And yea, original firmware has two modes. I can try this firmware again someday, i'll be in touch then.

-Jami

hogyoku, 2023-04-28:
Any updates on this ?