danielnilsson9/bbs-fw #98: New release?

MrHumanRebel, 2023-03-15:
Hi!

I'm interested in a new release of the FW in the current lastest state. (To have the "Keep current" option available)
Are you planning to build in the near future or not really?
This project is cool tho. 🥇 
Thanks in advance!

silenec, 2023-03-25:
I guess you can get 1.3.0 here until there's an official release: https://endless-sphere.com/sphere/threads/bbs-fw-open-source-firmware-for-bbshd-bbs02-controller.117092/page-7#post-1753272

closed 2023-03-31 by MrHumanRebel

danielnilsson9, 2023-04-01:
Official 1.3.0 release is now available. No significant changes from what was previously built from master branch.

silenec, 2023-04-01:
"There is an error in XML document (2, 2)" when opening saved 1.2.0 xml configuration in 1.3.0 config tool. Is it supposed to work?

danielnilsson9, 2023-04-01:
No sorry, there is no backwards compatibility on that document implemented. There is backwards compatibility when reading firmware from controller. So you can connect to your bike running 1.2.0 using configuration tool from 1.3.0 release and read out the config and save a new file.

silenec, 2023-04-01:
Thanks! I asked because I have not flashed the motor yet but I have already created a profile :D No problem, I just opened both versions and manually typed in a 1.3.0 profile again.
Then I spent 2 hours actually flashing the firmware. It always failed after a few seconds of uploading the hex file but luckily it was just a bad USB cable. I found a brand new one and everything seems to work so far - the motor spins, throttle and PAS works, I can see the temperature, config tool works etc.
Just one thing I noticed - when you use walk mode and let go off the button, the motor actually spins up to higher rpms for a split second before slowing down and stopping.

danielnilsson9, 2023-04-02:
Sounds like a potential bug, I was unable to reproduce when I tried now. Please open a new issue and try to provide some more details and maybe post your config. Is the behaviour the same independently which assist lever you were in before engaging walk assist? Do you have this issue even if you are in assist level 0?

silenec, 2023-04-02:
I quickly tried it yesterday after I flashed bbs-fw from stock firmware with the chain off the bike (to be safe) and then with rear wheel off the ground with the chain. You probably might not even notice it with the bike moving on the ground but there definitely was a clear and audible difference vs the stock firmware. I still have to take the bike for my first ride with bbs-fw so I will report after that.