danielnilsson9/bbs-fw #100: No power and battery problem with 36V 500W

Henriclay, 2023-03-20:
Hi, i had a bad adventure that might be helpful for the future. 

I have the Bafang BBS02B 500W 36V. 
I have these versions of the MCU : STC15F2K60S2   F/W version: 7.2.5S
So, as wrote on the instructions, it is compatible. 
So i installed the 1.2.0 firmware doing everything is told here (https://github.com/danielnilsson9/bbs-fw/wiki/Flashing-the-Firmware). And after that, on the display the battery icon was flashing (an empty battery icon appears and disappears) and i had no power, but the speed sensor worked.
So i was annoyed, and finally found a new hex file here : https://electricbike.com/forum/forum/main-forum/trouble-shooting/122397-firmware-for-36v-bafang-midtmotor 
And then it worked. But i'm back to a factory version of the firmware which was not the purpose but it's better than nothing.

I think that is does not work for the 36V 500W. What i have not done is that i didn't put back all the original configurations informations like it's wrote here : https://github.com/danielnilsson9/bbs-fw/wiki/Upgrading-Firmware-Version. I didn't because i never changed anything there, i just needed to flash the firmware because i have an intermittent error 30H that i cannot explain.

Maybe it's the problem. If danielnilsson9 know what happened, it could be great. Thx

Gonespa, 2023-03-20:
Hi,

Regarding the 1.2.0 FW, did you configure the low voltage cutoff value after flashing the FW unsing the config tool?. Looks like the FW is just considering whatever voltage value your battery currently has as below that level.

Henriclay, 2023-03-20:
Like i said, i didnt change anything in the config tool. I understand that maybe it's just possible that i needed to change that, putting a good voltage in it. 
I have stock parameters so maybe i need to do that. There is a lot of parameters and i dont want to do the wrong things. I dont know which max voltage and which low voltage cutoff. 
I'm gonna try it because the 30H error came back
Thx for you response,

danielnilsson9, 2023-03-20:
Default configuration is for 48V battery. If you plan to run any other voltage you need to set low voltage cutoff and max battery voltage parameters or it will not work.

You can set:
LVC: 28
Max voltage: 42

As for error 30H, that is usually ha hardware error, you most likely have a fault in your controller or cable. Changing firmware is not likely to solve that.

Henriclay, 2023-03-20:
Okay so i tried it. 

But, the configuration tool doesn't work with my controller... STC-ISP works well and other configurations tools are working, but not this one. It detects the USB port, but when i click on "connect", nothing happens, even if i disconnect and reconnect the battery. So i can't change LVC or Max Voltage.
What i've done is that i put the new firmware (didnt change anything, the config tool was still not working), and then i tried another configuration tool, but this one detects it like a 48V controller, and just prevent me to put a lower LVC than 38 (i think i could put 38-44 something like that).
So maybe, a solution is to create a firmware file adapted to 36V, or maybe i should have used another configuration tool that doesn't limit anything like the one i used. 

I know for 30H, but i don't find the problem, because the problem is intermittent, even if i don't touch anything sometimes i have the problem sometimes not. I heard it could be the display, but i don't know. So i just bought a new motor online because i had already planned to have a spare bike, and it will help me to understand what is the cause of the 30H.

danielnilsson9, 2023-03-20:
The only configuration tool that works with this firmware is the one that comes with this firmware, you cannot use any other configuration tool.  Documentation is here https://github.com/danielnilsson9/bbs-fw/wiki/Configuration-Tool

If you were able to connect with another config tool after flashing this firmware then I don't believe the flash was successful because that should not be possible. 

Display, cable or controller, there is likely a bad connection somewhere if controller works otherwise.

Gonespa, 2023-03-20:
> Okay so i tried it.
> 
> But, the configuration tool doesn't work with my controller... STC-ISP works well and other configurations tools are working, but not this one. It detects the USB port, but when i click on "connect", nothing happens, even if i disconnect and reconnect the battery. So i can't change LVC or Max Voltage. What i've done is that i put the new firmware (didnt change anything, the config tool was still not working), and then i tried another configuration tool, but this one detects it like a 48V controller, and just prevent me to put a lower LVC than 38 (i think i could put 38-44 something like that). So maybe, a solution is to create a firmware file adapted to 36V, or maybe i should have used another configuration tool that doesn't limit anything like the one i used (didnt had time for that and maybe its not recommended).
> 
> I know for 30H, but i don't find the problem, because the problem is intermittent which is rare, even if i don't touch anything sometimes i have the problem sometimes not. I heard it could be the display, but i don't know. So i just bought a new motor online because i had already planned to have a spare bike, and it will help me to understand what is the cause of the 30H.

Config tool sequence to connect is a bit weird but works:
Run the program with the cable connected to the PC. (So that the COM port to which it is connected shows).
Select the COM port and click connect ensuring you have your bike THE POWER OFF ( disconnected from your battery, otherwise it wont connect)
Program will start to try to connect.
Then power on your bike( plug your battery). It should then connect.

Henriclay, 2023-03-21:
I swear i've done everything told. I did the power off/on thing. And this procedure worked for flashing (with check MCU et download/program, i disconnect and reconnect and it works. Also did everything told). I know it's weird i didn't understand either. 

I think the flash worked because the other configuration tool told me that it was a 48V with another firmware version. If i have time i can post some screens here about that, maybe after receiving my new motor. 
I only screened before flashing (see below) but all these informations changed. It's possible that it shows some infos but it's not fully compatible. 

[image]

dav0000000, 2023-03-21:
Did you set the system clock speed when flashing the firmware to 20,000?

dav0000000, 2023-03-21:
[image]

Henriclay, 2023-03-21:
Thx, i will probably try that, i don't remember what i put but the controller was still working after the flashing. 
For now, it's working again and i don't have an error 30H but an error 25H. Error 25h is, apparently, a problem with gear sensor or brake sensor. I don't use these. I've never had the 25H. But with the 25, the motor works.  After all my flashing tries, i changed the parameters with the other config tool, and it changes a lot of things when riding. 
I think the 30H will come back. I will eventually try another firmware flash if it reappears. 
As we say in french, "perdu pour perdu...".

danielnilsson9, 2023-04-01:
I didn't know there existed a 36V version of the BBS02. The issue here was likely the LVC not set correctly but there could be other parameter changes in the original 36V firmware that should be set on the other motor control microcontroller.
Thanks for the links to the 36V firmware, I will try to investigate at some point.

danielnilsson9, 2023-04-27:
No update, closing.

closed 2023-04-27 by danielnilsson9