danielnilsson9/bbs-fw #103: public mode - private mode

SimonTheWay, 2023-04-01:
It would be interesting to find a way to switch from public mode to private mode.

If I turn the bike on normally, the bike will run in street legal mode (250W max continuous power, 25km/m max speed)

If I turn the bike on with a trigger (it could be both brake pushed) the bike will run in private mode (full power available)

#feature

naldoxable, 2023-04-02:
It is already implemented, check the wiki please.

danielnilsson9, 2023-04-12:
I think this issue refers to an additional option in which a switch to the other mode can occur. There is currently three configurable options (Eco Mode, External Lights or PAS0 + External Light).  This would be a fourth option (Brakes Active on Startup), when this mode is selected it will however not be possible to change mode without cutting power to controller.

I will consider this for next release.

manusalgam, 2023-04-13:
It would be more usefull if you can select a PAS level (for example, 7, not only 0), and turn on/off lights or activate walk mode or whatever, or maybe braking 5 times in 10 seconds in PAS0 when no speed registered..

Why not let power users choose what input/sorts of inputs (+,-,brake,PASX, etc.) to change between modes with AND, OR, XOR expressions and time factor? the idea is having a custom gangway that only user knows

danielnilsson9, 2023-04-13:
Yeah I guess I could see the use in your suggestion but that is a considerable amount of effort to implement and I will not invest the time in that. 

Easiest way if you want something custom is just to make the specific change yourself in the code and compile your own version. Then no one will know your secret code.

danielnilsson9, 2023-04-13:
I decided to add the rest of the PAS + Lights options along with the Brakes @ Power On option.

DonJohnLong, 2023-04-22:
More options for switching to sport mode would be very welcomed. i rather buy you some beers than paying fines. best regards

closed 2023-08-01 by danielnilsson9