danielnilsson9/bbs-fw #104: Battery SOC compensated Keep Current (%)

silenec, 2023-04-02:
I probably get around 15% less power near LVC compared to fully charged battery due to voltage drop.

The idea is to gradually (and optionally) increase Keep Current (%) by x % as the battery voltage goes down to keep the PAS power output more consistent.

So for example we start with the default value of Keep Current (60%) with fully charged battery and ramp up to ~75% Keep Current near LVC. Is this doable? Good or bad idea? 

There are more ways to implement this but this way the current would not get above the Max Current (%) setting because Keep Current (%) is usually set lower than that.

danielnilsson9, 2023-04-12:
Yes, I have been a bit annoyed by this too lately. It's not really a keep current thing, this affects all types of pedal assist since current (A) is used for motor control but Power (W) = Current (A) x Voltage (V).

So there should ideally be a calculation to compensate for voltage drop when calculating motor current percentage to apply.

I will tag this for next version.

silenec, 2023-04-12:
Yes it does affect all assists but keep current is where the power drop is a) the most annoying and b) easiest on the battery to provide more current.
I don’t really care if the acceleration (hardest on the battery) up to a steady keep current speed is a bit slower or if I get a bit less power in the same throttle position - I just apply more throttle to compensate.
I am sure whatever you think is best will be ok. Thanks

danielnilsson9, 2023-04-12:
Yes exactly, there is no point in compensating throttle power. But doing it globally after applying pas/cruise power would be an easy fix.

zeynelozturk, 2023-04-12:
> Yes, I have been a bit annoyed by this too lately. It's not really a keep current thing, this affects all types of pedal assist since current (A) is used for motor control but Power (W) = Current [...]

I was also thinking about that, probably there is a reason that watts isn't used instead of current (amps)?  I'm also curious if watts used for levels (instead of current), the resulting power would be similar between ~58.8v fully charged battery and 48V one for each PAS level.

danielnilsson9, 2023-04-12:
> I was also thinking about that, probably there is a reason that watts isn't used instead of current (amps)? My guess is voltage sags as you use current and it would maybe make power inconsistent?

Amps are used for BLDC control algorithm. I guess it was easy to continue to use it since it required less calculations. Power fluctuations could be a problem, but the voltage could be slowly filtered to avoid that I guess.

> I'm also curious if watts used for levels (instead of current), the resulting power would be similar between ~58.8v fully charged battery and 48V one for each PAS level.

Yes, that is the idea, except for high assist levels though, when you hit 100% current after adjusting (mostly applies to BBS02/BBSHD, don't think many people ride 100% PAS on BBSHD).