danielnilsson9/bbs-fw #105: PAS + Throttle + MaxSpeedPercent = problem

silenec, 2023-04-02:
This works fine:
`<AssistLevel Type="Pas Throttle CadenceOverride" MaxCurrentPercent="12" MaxThrottlePercent="80" MaxCadencePercent="100" MaxSpeedPercent="100" TorqueAmplificationFactor="0" />`

This is unusable:
`<AssistLevel Type="Pas Throttle CadenceOverride" MaxCurrentPercent="12" MaxThrottlePercent="80" MaxCadencePercent="100" MaxSpeedPercent="20" TorqueAmplificationFactor="0" />`

When I pedal and apply throttle at the same time, it goes something like this:
1000ms of 0% power, 200ms of power, 1000ms of 0% power, 200ms of power and it repeats until I let go of the throttle or stop pedaling. Throttle only or PAS only works fine with this configuration.

Config file attached (rename to .xml): 
[bbsfw 1.3.0.txt](https://github.com/danielnilsson9/bbs-fw/files/11132768/bbsfw.1.3.0.txt)

Also when I use walk assist (any PAS level or configuration) and let go of the button, motor goes like: spinning at 3% power, let go of the button -> 100ms of 25% power -> 0% power. But that could be another issue. Stock FW goes like 3% -> 0% with no power spike.

Apart from this everything seems fine with my BBS02B. Runs great. Pretty responsive even with 3 A/s Current Ramp and 90 degree Start Delay. I even tried cruise control, it was fun.

Maybe if I could have one general safety improvement request: **when pedaling backwards, disable throttle**. This could help in a malfunctioning throttle / damaged wiring situation if you run with no brake sensors like I do. I had it happen once with stock FW where my bike suddenly went 100% throttle with no throttle input. Not fun and any additional way that could help you stop the bike would be helpful.

danielnilsson9, 2023-04-12:
1. I will investigate, sounds like an edge case bug. 
2. I think I know what this is, I might have seen it, will try to fix.
3. I'm not to keen on implemented that, would have to be on yet another config option since disabling power when pedaling backwards on throttle would be really annoying when riding offroad where you constantly adjust your pedal position by pedaling backwards. I guess it could be if you pedal 2+ revolutions backwards you could disable though.

danielnilsson9, 2023-04-12:
I'm not able to reproduce your first issue, does it occur when you hit the 20% speed limit (20kph if you use the default max speed of 100kph)? Is your speed indication on the display correct or is it jumping too?