danielnilsson9/bbs-fw #106: Unused icon on the LCD

silenec, 2023-04-08:
There are 2 small icons on my C961 display - person on a bicycle (PoB) and person walking (active in walk assist mode with stock FW).

If I remember correctly from my last ride (not 100% sure now), then only the PoB icon shows with BBS-FW and does not change.

Now if someone could figure out how to trigger the person walking icon independently (if possible), it could be used to indicate selected Operation Mode (Standard / Sport) or some other things like cruise control (de)activated, throttle enabled / disabled, motor temperature getting high etc.

https://youtu.be/L03m1jq5QDU?t=78

danielnilsson9, 2023-04-12:
Are you sure that icon is just not showing "not in walk assist mode". E.g. always on when an assist level is selected? Or when does it turn off? I doubt there is any way to trigger that icon independently, it is probably a display firmware thing, not exposed on the serial protocol.

silenec, 2023-04-12:
The bicycle icon turns off and person icon turns on when in walk assist as on the YT video, yes. Thing is, I do not remember this happening with bbs-fw in walk assist. Bicycle icon stayed always on and person icon always off. At least that is what I remember but I could be wrong because I did not have a chance to verify yet. I will when I fix my bike or someone else can try and report.

closed 2023-04-17 by silenec