danielnilsson9/bbs-fw #107: BBS02B Shift sensor

kb79000, 2023-04-09:
[bbsfw.log](https://github.com/danielnilsson9/bbs-fw/files/11185643/bbsfw.log)
Hello everyone A big smile following the firmware test having a progressive accelerator is great compared to the big of the original firmware.
I open this incident to report my problem with the Shift Sensor which does not cut the current when changing speed and unfortunately causes cracking in the cassette😬
What do you need to try to debug the incident? 
Sorry for my approximate English 
Thank you to all the contributors for this great project 👍

EigenMania, 2023-04-10:
First check that "Shift Sensor" is checked in the config (and that it is correctly flashed to the motor). If there is no power cut at all, it could be faulty hardware (try with another shift sensor if available). If the power cut does occur, you can try increasing the interrupt duration, or decreasing the current threshold, or a combination of both.

kb79000, 2023-04-10:
> First check that "Shift Sensor" is checked in the config (and that it is correctly flashed to the motor). If there is no power cut at all, it could be faulty hardware (try with another shift sensor [...]

There is a cut but it is so small that when shifting small speeds with a lot of power it causes what a strange noise on the cassette
I tried a lot of settings like the Shift Interrupt Duration at 2000ms and the Shift Current Threshold at 0% without improvement do you also use it on a BBS02B? 
Thx

EigenMania, 2023-04-11:
I tested on BBSHD and BBS02B and on both systems I was able to change the interrupt duration and current threshold. I will test on the 1.3.0 release to confirm if it still works or not.

kb79000, 2023-04-11:
> J'ai testé sur BBSHD et BBS02B et sur les deux systèmes, j'ai pu modifier la durée d'interruption et le seuil de courant. Je vais tester sur la version 1.3.0 pour confirmer si cela fonctionne toujours ou non.

Adjustment is possible on my BBS02B but has no effect despite different settings, I await your feedback 🙂

danielnilsson9, 2023-04-12:
There is a type in the code on line 31 in fwconfig.h which disabled shift sensor support on BBS02... Sorry about that, will release a patch.

kb79000, 2023-04-12:
> Il y a une faute de frappe dans le code à la ligne 31 dans fwconfig.h qui désactive la prise en charge du capteur de décalage sur BBS02... Désolé, nous publierons un correctif.

Ok thank you 🤗 @danielnilsson9 you think the correction will be made during the week ?

manusalgam, 2023-04-12:
Complied and working now properly in my BBS02B, shift sensor appears to work (I'll take a run and test, I've done using throttle but without riding)

kb79000, 2023-04-12:
> Complied and working now properly in my BBS02B, shift sensor appears to work (I'll take a run and test, I've done using throttle but without riding)

How to compile with the modification of Daniel? Sorry I'm a beginner😚

manusalgam, 2023-04-12:
I may confirm it works after a short ride (my shift cable isn't too thick and i've to fix, sometimes doesn't work but is not a FW problem anymore, happened exactly same in original FW), thanks a lot Daniel!!!

danielnilsson9, 2023-04-12:
Great! Yeah will try to update the thread someday..

Big thanks to @EigenMania has done the work on the tunable shift sensor feature which was part of this release. I just managed to screw it up when integrating with my other changes.

1.3.1 has been released, I will leave this issue open for a few weeks for anyone coming here with this issue.

kb79000, 2023-04-12:
I have the error 30 hours after flashing and reset eprom in the configuration tool an idea?🤔

danielnilsson9, 2023-04-12:
Error 30 means communication between display and controller is not working, can be a lot of things.

- Broken display cable
- Broken controller cable
- Incorrectly flashed firmware
- Dead controller (not likely if it was just working)

When separating display connector you have to be very careful not to pull one the cable or you could easily damage them.

1. Try flashing firmware again (remember 20Mhz!). If unable -> display cable might be broken on controller side.
2. Firmware flashed but display still showing error 30 -> try flashing original firmware (18.432 MHz!).
3. Original firmware flashed -> display still showing error 30 -> likely damaged display cable connector.
Try another display if you have one.

kb79000, 2023-04-12:
> L'erreur 30 signifie que la communication entre l'affichage et le contrôleur ne fonctionne pas, peut être beaucoup de choses.
> 
> * Câble d'écran cassé
> * Câble du contrôleur cassé
> * Firmware mal flashé
> * Contrôleur mort (peu probable s'il fonctionnait simplement)
> 
> Lorsque vous séparez le connecteur de l'écran, vous devez faire très attention à ne pas tirer sur le câble ou vous pourriez facilement les endommager.
> 
> 1. Essayez à nouveau de flasher le firmware (rappelez-vous 20Mhz!). Si impossible -> le câble de l'écran est peut-être cassé du côté du contrôleur.
> 2. Le micrologiciel a flashé mais l'écran affiche toujours l'erreur 30 -> essayez de flasher le micrologiciel d'origine (18,432 MHz !).
> 3. Le micrologiciel d'origine a flashé -> l'écran affiche toujours l'erreur 30 -> le connecteur du câble de l'écran est probablement endommagé. 
>    Essayez un autre écran si vous en avez un.

Do you think there is an error in the flash on this image?https://ibb.co/mSKkj3m

danielnilsson9, 2023-04-12:
That looks like a successful flash!

kb79000, 2023-04-12:
> Cela ressemble à un flash réussi !

Ok I think my screen connector is HS😭

kb79000, 2023-04-12:
> Error 30 means communication between display and controller is not working, can be a lot of things.
>
> * Broken display cable
> * Broken controller cable
> * Incorrectly flashed firmware
> * Dead controller [...]

I would like to reflash in 18.432MHz but I can't find the original Bafang firmware for a 48V battery

Supertas, 2023-04-12:
I don’t know which firmware is this. But perhaps its help.

https://california-ebike.com/pages/updating-bafang-firmware

closed 2023-04-17 by danielnilsson9