danielnilsson9/bbs-fw #108: BBS02 Config Write Error

MrHumanRebel, 2023-04-13:
Any ideas? 
Tried connecting, rebooting.
The firmware flashing process was flawless.

[image]

kb79000, 2023-04-13:
When you are logged in with the screen do you have an error code?

danielnilsson9, 2023-04-13:
No idea what is causing this, sounds like some interference, it works correctly if you are not charging the battery?

MrHumanRebel, 2023-04-13:
Yes indeed! If the battery is not charging, it works fine, I even took a short trip with the new FW.

danielnilsson9, 2023-04-13:
Interesting, only thing I can think of it the charger producing a ripple which is interfering with the serial lines which runs in the same cable and could easily induce a voltage (noise) on the rx/tx lines given the length of the exposure and no shielding.

(B+ and B- runs inte main harness and display cable all the way up to the display)

Not likely to be a firmware bug in my opinion

MrHumanRebel, 2023-04-13:
That would be a reasonable explanation to this, I have no experience in the cable routing inside these motors. 
It may be good idea to include this finding to the documentation, to warn potential new users doing stuff to the controller while charging to avoid firmware corruption.