danielnilsson9/bbs-fw #110: Feature request: Display view upgrade

MrHumanRebel, 2023-04-13:
It would be nice to configure the default bottom screen view from these options:
- Current rolling (km)
- Current power (W)
- Motor temperature (C)
- ODO (km)

For example if you want to constantly view the current power instead of the factory default current rolling.
This would be only configurable throught the config tool, and not on the device itself to reduce complexity.

Another thing I might see useful, to be able to modify the time that takes the display to return to the default view when cycling through the menu. For example to change it to 10-15 seconds to be able to read the information properly.

Are my ideas feasable?

Thanks in advance!

danielnilsson9, 2023-04-16:
This firmware is for the controller and only provides data for display. Sounds like you are looking for a display firmware replacement, I'm not aware of any such projects for the bafang display/protocol.

MrHumanRebel, 2023-04-16:
Sorry, I did not know that this cannot be changed through the motor FW, I just thought it would be a nice thing to have. 
Thanks for letting me/others know!
Have a nice day!

closed 2023-04-16 by MrHumanRebel