danielnilsson9/bbs-fw #112: FR: SpeedLimit for Throttle per Assist-Level in PAS mod

Supertas, 2023-04-15:
Hi Daniel,

i test your firmware with an BBSHD today the first time. Very nice first impression.

I have a feature request or i doesn’t find the settings:
In Germany is throttle only allowed, when is not faster then 6km/h. But with PAS is allowed until 25km/h. I will use the throttle as a starting helper until 6km/h. I found only the power limitation for the throttle but not an speed limit.

I need a speed limit for the throttle per assists level.

Thank you for the very great project.

Thomas

danielnilsson9, 2023-04-16:
I see, I will think about how to add this, a speed limit option per assist level is a bit problematic. I guess in your use case as soon as you touch the throttle the bike should stop providing power if your are above 6km/h even if you are still pedaling?

A global option might work which could limit speed for throttle on all "Standard" assist levels?

Supertas, 2023-04-16:
A global option is ok for me, too.

yes over 6km/h I will pedaling with motor help until 25km/h.

Thx, l send a small donation for you.

Thomas

kb79000, 2023-04-16:
Hi yes this option is essential for the standard mode for Europeans because we have crappy regulationsHi yes this option is essential for the standard mode for Europeans because we have crappy regulations 😅

DonJohnLong, 2023-04-22:
I'm also interested in this becaus here in Switzerland you can use the trottle up to 20 km/h legaly.  A global trottle limit in standard mode would also be fine for me. 

Thank you so much for this project, i will keep sending donations for every new version.

kb79000, 2023-05-29:
Hello The accelerator limitation at 6 km/h is planned for 1.4.0 ? 
When will the new version be published ? 
Thanks for the work 🙂

danielnilsson9, 2023-05-29:
Yes its planned for 1.4.0. I was working on it this evening but I had missed DonJohnLongs comment which makes things a bit more complicated so I will have to start over. I can not really give you an eta as I'm not regularly working on this, only when there is time/energy.

kb79000, 2023-07-07:
Hi for my country the top is map the walking mode with the throttle

kb79000, 2023-08-09:
[bbsfw.txt](https://github.com/danielnilsson9/bbs-fw/files/12304687/bbsfw.txt)
Hi @danielnilsson9 I share my configuration with you, with the modification you made for the throttle clamp, I set it to standard levels at 6%.
I may not have fully understood how the option works, but setting the pitch to 6% works but also seems to be limited to 6%.

closed 2023-08-26 by danielnilsson9

emeikei, 2023-08-29:
Just tried this and I'm not sure... :/

Throttle alone:  Global Speed Limit applies 👍
Peddle alone: Assist Level limits apply 👍
Throttle and peddle: Global Speed Limit applies seemingly 🤔

In the UK/EU, the throttle only needs to be limited to 6kmh when used independently.  When peddling, the throttle can be used as normal up to the 25kmh speed limit.  Even if PAS is disabled, so long as the peddles are turning, the throttle is allowed.

So basically I think the Global Speed Limit should only be applied when the peddles are not turning.  If the peddles are turning then this limit/feature should be ignored :)

danielnilsson9, 2023-08-29:
I'm not sure I understand. 

If we make it behave as no global speed limit is configured, then throttle will override PAS when requested current from throttle press is larger than the configured PAS current giving a smooth transition from PAS to throttle. The opposite is also true, PAS overrides throttle when configured PAS current is greater than requested throttle current.

Then we implement global throttle speed limit as follows:
* If pedals are not turning throttle can be used up to 6km/h
* If pedals are turning throttle can be used and overrides PAS when requested throttle current is greater than configured PAS current.

If that is what you are saying then that is the problem I tried to describe.

Given the rules above. If you are walking with you bike with 4% throttle current and limited to 6kph, you hit a stone with you pedal and it starts spinning. PAS current is configured to 30%. Then the global throttle speed limit will not apply since pedals are spinning and PAS current was configured to 30% which is more than 4% and the bike will take off.