danielnilsson9/bbs-fw #114: Potential bug: Speed sensor loses tracking

MrHumanRebel, 2023-04-23:
Recently rode the bike with the latest 1.3.1 version for a long run, I noticed around every 10-15 minutes the speed sensor data is not shown properly for a few seconds. 
For example cruising at constant 35 km/h and sor a few moments firstly a lower speed is shown like 15-20 km/h and then a spike up to the 50s, then it is back to normal. 
Might be a sensor pulse filtering issue, but I'm just guessing at this point, I have no more information about his anomaly. 

The speed sensor and the magnet is located perfectly to each other, and working (and so blinking) perfectly fine.
This is with a BBS02B.

danielnilsson9, 2023-04-23:
No idea. I think there is one other guy on a BBS02 who had similar issue too along with a lot of other issues. Haven't seen a this on my BBSHD or TSDZ2. I can leave the issue open but if this was a common firmware issue on the BBS02 I would have expected quite a few reports.

I own no bike with BBS02 so I have no experience of how it runs on that controller, support was added blindly by me and then tested by others.

kb79000, 2023-04-23:
Hi, i did not notice this problem on my bbs02b

silenec, 2023-04-23:
Yeah I did notice this on my last longer ride with road speed limited PAS levels. The motor randomly cuts power for no apparent reason and spins up again while the speed on the display goes up to 50s. Not very often though as you say so it is hard to diagnose.

ronmicro, 2023-04-25:
> Yeah I did notice this on my last longer ride with road speed limited PAS levels. The motor randomly cuts power for no apparent reason and spins up again while the speed on the display goes up to 50s. Not very often though as you say so it is hard to diagnose.

My bbs02b with v1.3.1 has this same problem of random power cuts, it happens about 2 or 3 times in a 10 mile ride

danielnilsson9, 2023-04-25:
Alright, definitely sounds like a firmware issue then. Was anyone of you running 1.2? Was the issue present there too or is it new in 1.3?

MrHumanRebel, 2023-04-25:
> Alright, definitely sounds like a firmware issue then. Was anyone of you running 1.2? Was the issue present there too or is it new in 1.3?

Personally, I started using this firmware from 1.3.1 so I have no information regarding this.

ronmicro, 2023-04-29:
I started from 1.3.1 too but i just tried version 1.2.0 on a long ride and it still happened a couple of times

Larswagner1987, 2023-04-29:
> Ich habe auch mit 1.3.1 angefangen, aber ich habe gerade Version 1.2.0 auf einer langen Fahrt ausprobiert und es ist immer noch ein paar Mal passiert

that's a shame, do you have any idea what could be wrong, when does it happen, is there a trigger or did you notice something?

Gonespa, 2023-04-29:
Hi,

I Have 1.2.0 too and noticed the same. I though this was related to the fact that i have max current set to 33 amps in a bbs02 and it was hitting some kind of extreme condition but I see it might be due to other thing. In my case, I noticed that behaviour when locking the rotor and demanding max current for a few seconds. I managed to see an error coming in the log.

Larswagner1987, 2023-04-29:
I also suspect that it has something to do with the current, every time the motor is supposed to give current it happens, on the road it's just not a problem but as soon as there is an increase it starts as soon as he has to give more current then he interrupts 1-2 seconds and then the line comes back, I have a bbs02b motor 48v 750Watt

dav0000000, 2023-04-29:
Are you guys using pas or throttle when it happens and if you are in SPORT, is it still in SPORT after the problem?

Larswagner1987, 2023-04-29:
ich benutze nur pas und die normal , also kein Gas oder Sport

Gonespa, 2023-04-29:
I have only noticed this when using the throttle. I am not 100% sure about the mode, but I would say thay the mode is not changing. 
If it happened I would probably not notice it since I have configured the "standard" mode as full power mode (and the one I use 99% of the time0 and "sport" mode as a tamed down one.

Larswagner1987, 2023-04-29:
I have no gas in use, I tried again to generate the error and it only comes when the engine draws power, then the power goes away and comes back then it's fine and at some point it starts again, it's weird, it can  but just look at my display and see how the watt reading goes to 0 in time and then is full again.  must be a small thing, but where only in the log is there nothing in it for errors.

danielnilsson9, 2023-04-29:
@Gonespa Probably not related to this issue unless others also have their current set that high. It's been reported before that it's not possible to set current above 25A for BBS02 because some other hardware protection is triggered outside the control of this firmware, that is probably what you are seeing. Set max current to 25A and it should work normally.

Larswagner1987, 2023-04-29:
@danielnilsson9 I have mine set to 23a and it can be 25a, could it be because the engine had the misfires as a result???

Gonespa, 2023-04-30:
> @Gonespa Probably not related to this issue unless others also have their current set that high. It's been reported before that it's not possible to set current above 25A for BBS02 because some [...]

Agreed, I only mention it since the described behaviour looks exactly as I saw. I mean, behaves just like when the motor control micro returns an error: power disconnects completely for 1 or 2 seconds and the resets. Looks to me it is returning some kind of error. I was thinking, is it possible to save in permanent storage "the first error" returned, so that it can be retrieved like a car MCU does with the app?. 

Anyways, other than that extreme condition, I also noticed this a few times when using the bike in much less stressing conditions.

Larswagner1987, 2023-04-30:
@danielnilsson9 The only thing I have done now is connect the laptop and start and then the error comes, but it only shows Motor target current changed to 0% for 1 second and then it goes back to normal, and once it says Failed to read voltage from motor controller, everything was fine so far.

Larswagner1987, 2023-04-30:
@danielnilsson9 

I've tried everything now and it's a pity that this error is there, as soon as the laptop is connected it doesn't come up anymore as soon as it's not connected it comes up which is strange, the cables are new.  I tried different settings but it didn't work, I don't know what the problem is anymore, sometimes it's there immediately, sometimes it's only there after 4 km, then it doesn't work at all, it just breaks off and then the performance comes back after 1-2  Seconds, I've got the original bbs02 version back on it now, if you find the bug and then have a new version I'll try it.

danielnilsson9, 2023-04-30:
I see, yes you are probably better off with the original firmware. These issues are not likely to be fixes as I don't have a bbs02 and are not doing any work on that motor. So unless someone else decides to do some investigation here it will probably stay this way.

Larswagner1987, 2023-04-30:
@danielnilsson9 

ok shame, like I said, it's great except for the error, I can only send you a bbs02 controller or the original hex file, can you do something with it then?

silenec, 2023-04-30:
I got back from another ride and I can say this bug appears way more often and / or is more noticeable when going uphill.
I can pretty much predict some situations where it happens. It almost always happens when approaching a hill.
- when road speed goes down and current goes up, the motor stops and starts again like 90% of the time
- it rarely happens when road speed goes up and current goes down (going downhill)
- it happens less often when going at constant speed in lower PAS levels
- it happens less often when road speed is higher (= more pulses from the speed sensor) even with stronger assist level
- it is probably the same problem I mentioned in https://github.com/danielnilsson9/bbs-fw/issues/105
- when going uphill in PAS5 (PAS + MaxSpeedPercent, config file in the link above), I got the motor to constantly skip even without throttle - previously I though it only happened with PAS + Throttle + MaxSpeedPercent but that is not the case
- it does not happen when using throttle only (no PAS)
I think we should test some bare minimum firmware - no temperature monitoring, no throttle, no keep current, no push walk, no ramping anything that is not necessary, maybe even no speed sensor if possible etc. and see when it stars working properly.

Larswagner1987, 2023-04-30:
@silenec then try it and let me know, I definitely need the speed sensor and the shift sensor too, I don't need the rest like the gas, I hope that the error trigger is found, the original version is really not nice to use there  this version is perfect.  thank you already 👍

danielnilsson9, 2023-04-30:
@Larswagner1987 Thanks but I have a BBS02 controller and firmware but diagnosing things that only occurs while riding on high power is hard with a controller on my desk.

@silenec That makes sense, I was never able to reproduce that issue, I have ridden > 100km since then using PAS and throttle on high power on my BBSHD but it's rock solid, I have not seen a single glitch. 

Weird thing is that there is not really any BBS02 specific code here in this firmware apart from pin assignments and motor MCU configuration. The motor MCU config was directly mirrored from the original BBS02 firmware. 

Are you seeing speed glitches on the display too?

silenec, 2023-04-30:
Sometimes I see speed glitches on my display when it happens but sometimes I do not yet the motor still stops. It is weird. Maybe something in the NEC MCU does not like how the new FW behaves? Maybe there are some limits like maximum change of current / % of requested power per second or something and the limits might be different for BBSHD? And when something goes over the limit, NEC MCU "resets" and starts again? I have no idea honestly.

danielnilsson9, 2023-04-30:
Ye maybe, I was thinking along these lines too as there is actually an error code reported from NEC MCU on BBS02 described here https://github.com/danielnilsson9/bbs-fw/wiki/NEC-Motor-Control-Protocol which I guessed was an overcurrent error but could be anything really. It's not exported to any error code sent to the display so you will never se an error when it occurs. Would be interesting to know if that error is set when this issue occurs.

NEC MCU shutting down doesn't explain the speed sensor reading issue though, since speed sensor reading is done by this firmware..

silenec, 2023-04-30:
Now that I think about it... I put new chain on an older cassette once and the chain used to skip. The motor somehow detected that and every time the chain skipped, the motor did exactly this - stopped and spinned up again. Same thing when my rear tire slipped on ice in winter - motor stopped and started again. Not sure how that worked exactly. That was a few years ago on stock FW. Might it be the "chain skip protection" (if there is one) getting triggered by glitchy speed sensor values? Is the speed sensor reading reported to the NEC MCU or not? I guess not so this would not make sense then... hmm

dav0000000, 2023-04-30:
I had speed sensor rise with a watchdog reset a while back. Only way I knew was SPORT mode always reset to STANDARD as the system restarted.

Larswagner1987, 2023-05-02:
Hello

have you tested everything? a friend of mine thinks it could be something with the magnets, because when you drive and the power is gone, he took a closer look and noticed that it could be in the pas leagues, but there could also be incorrect information, the second could be there it's all mirrored what's wrong with the current values, since the BBs motor does a mini reset and then it's full again, or could it be that the memory/ram is too full and that's why it's happening?

ronmicro, 2023-05-03:
I tried disabling the speed sensor in the config tool today but made no difference, still a lot of drops over a long ride

danielnilsson9, 2023-05-03:
Maybe its RAM related anyways, its the only thing I can think of that is different between BBS02 and BBSHD, BBSHD has 4kb of ram, BBS02 has 2kb and there is only like 4 bytes of free ram left when compiling for BBS02. As per my current understanding that should be ok as everything is allocated upfront but maybe I have misunderstood something or there is a compiler bug causing these issues.

I will fix issue #86 and see if that helps, it's worth a shot. I'm also considering trying to compile with the Keil compiler and see if that makes any difference.

MrHumanRebel, 2023-05-08:
> 

Thanks Daniel in advance for your hard work and patience debugging these stuff!

Larswagner1987, 2023-05-08:
@danielnilsson9 Thanks also from me and my wife, you are doing a great job, you are the best. Thank you thank you thank you.

silenec, 2023-05-11:
Meanwhile I went back to stock FW and the skipping is gone, no speed jumps on the lcd. So just to confirm, definitely not a hardware / sensor issue.

silenec, 2023-05-12:
I used the biktrix one named CRR1094824E010015.3_170815.hex
https://biktrix.stonly.com/kb/guide/en/52v-programming-for-bbs02-motor-K2gUq2nWAX/Steps/1132765
It is a "52V" FW but it works fine with my 12S battery (minimum LVC is 38V, same as with the stock-stock FW before I flashed anything).

danielnilsson9, 2023-05-30:
Thanks for testing! I figured it would not help while rereading all the comments in this thread. Must be something else. I'm gonna forward the error code from the NEC MCU to the display so we can know at least if that specific error is triggered and reported

Larswagner1987, 2023-06-05:
Hello, have you already found the error? so far everything is going well except for error 09, I would be happy to hear that everything works without problems, the software is simply perfect and is what you want, I can only say thank you to Daniel for his great work on it.

danielnilsson9, 2023-06-05:
> Hello, have you already found the error? so far everything is going well except for error 09, I would be happy to hear that everything works without problems, the software is simply perfect and is what you want, I can only say thank you to Daniel for his great work on it.

No I don't think it is resolved, see comment above by MrHumanRebel. The issue is still remains, fixing ram usage did not have any effect.

Larswagner1987, 2023-06-05:
> > Hallo, hast du den Fehler schon gefunden? Bisher läuft bis auf Fehler 09 alles gut, es würde mich freuen zu hören, dass alles ohne Probleme funktioniert, die Software ist einfach perfekt und entspricht Ihren Wünschen, ich kann mich nur bei Daniel für die tolle Arbeit bedanken.
> 
> Nein, ich glaube nicht, dass es gelöst ist, siehe Kommentar oben von MrHumanRebel. Das Problem besteht immer noch, die Behebung der RAM-Nutzung hatte keine Auswirkung.

Okay👍

MrHumanRebel, 2023-06-06:
Rode another 100+kms with my bike as I ride everyday to work and back. 
Funny story, some bug got into my eye and I looked down and saw the km/h jumping from 35 to 28.5 and back to 35. (the motor was going flawlessly as 28.5<35 so no dropout)
So that means the speed sensor spikes not only happen to the + direction to the 50s (then the motor stops obviosly 50>35), but also to the - direction.
I noticed this firstly today.

ronmicro, 2023-06-12:
This might just be a coincidence but i changed my display from a DM03 to an 850C and didn't get any drops at all on my last 2 rides

Larswagner1987, 2023-06-12:
how do you mean that no pot get???  I do not understand that .

ronmicro, 2023-06-13:
Sorry what did you mean by no pot get? it didn't translate well.
Maybe some displays randomly send a value that makes it skip the switch case statement in process_bafang_display_write_pas()?

ronmicro, 2023-06-17:
Changing that function a bit seems to have fixed it for my DM03 display, went 16 miles with no power cuts yesterday

silenec, 2023-06-23:
[ronmicro](https://github.com/ronmicro) sounds good, is it still working for you? Can you share the modification?

ronmicro, 2023-06-23:
Yes i've been on a few rides with no power cuts now.
This is in the extcom.c file line 721

```
static int16_t process_bafang_display_write_pas()
{
	if (msg_len < 4)
	{
		return KEEP;
	}

	switch (msgbuf[2])
	{
	case 0x00:
		app_set_assist_level(ASSIST_0);
		break;
	case 0x01:
		app_set_assist_level(ASSIST_1);
		break;
	case 0x0b:
		app_set_assist_level(ASSIST_2);
		break;
	case 0x0c:
		app_set_assist_level(ASSIST_3);
		break;
	case 0x0d:
		app_set_assist_level(ASSIST_4);
		break;
	case 0x02:
		app_set_assist_level(ASSIST_5);
		break;
	case 0x15:
		app_set_assist_level(ASSIST_6);
		break;
	case 0x16:
		app_set_assist_level(ASSIST_7);
		break;
	case 0x17:
		app_set_assist_level(ASSIST_8);
		break;
	case 0x03:
		app_set_assist_level(ASSIST_9);
		break;
	case 0x06:
		app_set_assist_level(ASSIST_PUSH);
		break;
	}

	return 4;
}
```

Larswagner1987, 2023-06-23:
@ronmicro  

Can you tell us exactly what you did so that there are no more failures? I would be happy if it finally works without the failures, thanks in advance.

ronmicro, 2023-06-25:
That is the edited function, if you compare the original i just prevented it from being able to default to ASSIST_0.
You could probably test this on v1.3.1 by just programming assist level 0 to be active.

Larswagner1987, 2023-06-25:
> Das ist die bearbeitete Funktion. Wenn Sie das Original vergleichen, habe ich nur verhindert, dass sie standardmäßig auf ASSIST_0 eingestellt werden kann. Sie könnten dies wahrscheinlich auf Version 1.3.1 testen, indem Sie einfach die Unterstützungsstufe 0 so programmieren, dass sie aktiv ist.

So enter a value at level 0 that it is active or am I wrong there?

aalm, 2023-06-25:
> So enter a value at level 0 that it is active or am I wrong there?

I think that would be rather non-optimal workaround, given you would have only one level of settings
working without the issue, and to fix that, rebuilding with the above change by ronmicro will be easier.

Btw., which display are you having this issue with?

danielnilsson9, 2023-07-04:
@ronmicro Wow nice find! Thank you for investigating. Must be either some message sent that this firmware can't parse or some interference on the lines causing this. There is no checksum on this specific message...

I will add this fix for next version

Larswagner1987, 2023-07-13:
where can you find the beta version or is it still uploaded??

danielnilsson9, 2023-07-28:
@MrHumanRebel Thanks for testing, I will merge the changes. It's more of a workaround than an actual fix.

There are basically two things that are equally likely to cause these issues:

- An unknown command is sent by display which this firmware will not be able to handle and that will mess up the rest of the communication for a short period.
- Electrical interference. Sounds plausible based on reports that it's more likely to occur when high power is requested. Battery power cable routed along the display harness cable could make this more likely to occur.

@Larswagner1987 There are no precompiled beta versions, you will have to build from source or wait for release.

Larswagner1987, 2023-07-28:
oh okay, then I'm curious when it will be published, or can someone create a file for me that I can upload it, I don't know how I can do it myself.

silenec, 2023-08-24:
I flashed 1.4.0 and the workaround for this issue seems to work great - BBS02 is usable now.
Speed reading still goes up and down sometimes so speed limited PAS levels / street legal setup cannot be really used because of quite violent power drops and restarts. Other than that I am very happy with it. Seems to work perfectly with no (100km/h) speed limits set.

Larswagner1987, 2023-08-24:
wie meinst du das ?? bitte mehr Details.

silenec, 2023-08-24:
> wie meinst du das ?? bitte mehr Details.

https://endless-sphere.com/sphere/threads/bbs-fw-open-source-firmware-for-bbshd-bbs02-controller.117092/page-15#post-1770959

Larswagner1987, 2023-08-25:
it turned out perfect thanks to everyone who helped 👍👍💪👍👍💪

danielnilsson9, 2023-08-28:
@silenec So the speed limit kicks in when the speed on display goes up? That would mean that it is not just a communication error but rather a computation error if the computed speed is wrong too.

Cabal83958, 2023-08-30:
BBS02 with 1.4.0. 
I have the weird readings too, both Watts and Speed readings. But the motor works perfectly so I don't really mind. I don't think it's electrical interference because I haven't changed the hardware at all and it didn't do that before.

[33 more replies]