danielnilsson9/bbs-fw #116: Weird battery level indicator thing

MrHumanRebel, 2023-04-26:
The battery had 3 bars when I started the ride and the immediatelly it dropped to 2 bars (around 40%) on the screen, what is a perfectly normal behaviour in my humble opinion. 
I rode 3 km-s in the city, on PAS 3 (I have it set up for max current 30%), after that I went out to the main road and started cruising home. 
The battery percentage was constantly showing me 2 bars, and I thought it is legit too since I only went around 5km-s aboat that time, so it should have been real. 
I was cruising fixed 30 km/h and then a car went in front of me so I stopped pedalling and braked. After this the battery started blinking as it was completely empty. 
It was not, still had 45.1V when I got home and connected the controller to my laptop. (so the previous around 40% indication was indeed right)
The battery max voltage has been set to my 1 year old battery, it can hold 53.4V and I set the minimum voltage to 43V, so at 45.1 the battery indicator should have not blinked.
I was able to get home (it was a very windy day though), the battery had charge in it so it was "fine", but because the ramp down mechanizm I was only getting 100-150 watts of power and that is also normal.
Not sure why this happened of what could cause this, just thought to let you know.

Thanks in advance, and for your work making this cool firmware.

danielnilsson9, 2023-04-26:
Battery state of charge estimation is a hard problem, this firmware makes no effort to get it right. 

This firmware applies a simple voltage based approach, the logic is as follows:

- Take battery voltage measurements when no power is applied for more than 2 seconds.
- Consider battery full at 0.98 * max voltage
- Consider battery empty at 1.05 * LVC
- Start power ramp down at 1.06 * LVC (measured under load!, you will hit LVC ramp down way before battery SOC considers your battery empty due to voltage sag under load)

I.e. in your case:
- LVC ramp down starts at: 45.6V
- Battery considered empty at 45.1 (while under no load!)
- Battery full at > 52.3V

What kind of battery do you have? 53.4 - 43 volt sounds like a really small range, I can see why the above logic gives a bad result for you.

There is sort of a bug, the percentage in calculation should be on the useable configured range and not fixed on the end values  because that gives a bad result on lower voltage batteries and especially in your case with a small voltage range.

I will rework that to something more sensible for next release, should give a better result on lower voltage batteries. The current values makes sense for a 52V (58.8V - 42V) but are not very good for yours.

MrHumanRebel, 2023-04-26:
I have a 48V 18Ah Pananonic cell battery, that is with the BBS02B. 
I did not want to have a 52V pack, I rather have a longer life expenctancy of my motor controller because the BBS02B-s are not really designed for 52V, they work but you know...
I have this battery voltage thing from some forum from a year ago when I bought this set.
[image]
According to this (if this picture is indeed correct) originally my battery had 54.6V, now it can hold only 53.4 after 1.5 years of use.
Thank you for planning to rework the calculations a bit.

MrHumanRebel, 2023-04-27:
Another idea that jumped in, maybe it would be better to seperate the different types of calculations by the battery voltage. 
You could select the battery V in the config tool, for example 36V/48V/52V. 
Would it make any sense? Or just stay at one logic fits for all idea.

kb79000, 2023-04-27:
Hello I do not know if my problem is linked on my BBS02B with 48V battery when I turn it on there is the battery gauge which remains red 3-4 seconds at startup

dav0000000, 2023-04-27:
Every battery is different and the same battery also changes over time. If you are trying to make decisions based on battery percent then you probably should be displaying voltage and learning what voltage means for your battery, bike, riding style and terrain. I know just by looking at my battery voltage how far I can go and where. Sometimes I make a mistake and curse as I ride the last bit home with no power.

MrHumanRebel, 2023-04-27:
Yes every battery is different and will change over time. 
I know it for sure as an IT engineer. But that is why voltage calibration exists in this firmware, to manually adjust the V over time and recalibrate for example every 2-3 months.
Not everybody has a voltage meter screen on their bikes, neither me. 
The goal would be not to give an exact battery %, just a reasonable approximation, and try tom avoid situations as described above. 
As @danielnilsson9 already defined how the current calculations work now, and said it is mainly for 52V batteries, that is why this "issue" mainly exists.  And I can't thank you enough @danielnilsson9  for the level at which you deal with such little things.

dav0000000, 2023-04-27:
The voltage calibration is because every controller is different. It is not related to the actual battery. Different temperatures will also introduce inaccuracies. SOC (state of charge) counting is the only accurate way to do smooth battery percent but has one problem. A partial battery charge must estimate the SOC count accurately and the only measurement for that is battery voltage/percent and we start to go around in circles. Full battery charge and no problem when you know the actual mah of your battery. A lookup table will work better than a MAP for battery percent but it will be different for each battery. That's my two cents worth. And yes Daniel, you are simply amazing. Your programming and problem solving is simply out of this world.

dav0000000, 2023-04-27:
Could this be sagged voltage creeping into battery.c where - motor_disabled_at_ms > BATTERY_NO_LOAD_DELAY_MS?

danielnilsson9, 2023-04-27:
@kb79000 Its a known "feature", 0% is reported until a reading is available. Will see. maybe I fix it too.

danielnilsson9, 2023-04-27:
Fix implemented in master for both issues but not yet tested on real bike

Not sure it's possible to fix the 0% battery shown after boot, some display seems to show 0% until they have retrieved the first reading. But should be faster now at least.

kb79000, 2023-04-28:
> Correction implémentée dans master pour les deux problèmes mais pas encore testée sur un vrai vélo
> 
> Pas sûr qu'il soit possible de réparer la batterie à 0% affichée après le démarrage, certains écrans semblent afficher 0% jusqu'à ce qu'ils aient récupéré la première lecture. Mais devrait être plus rapide maintenant au moins.

The release of a new version to test your modifications? Thank😉

manusalgam, 2023-05-09:
> Fix implemented in master for both issues but not yet tested on real bike
> 
> Not sure it's possible to fix the 0% battery shown after boot, some display seems to show 0% until they have retrieved the first reading. But should be faster now at least.

In my display (750C BLE, BBS02B, 48V battery), sometimes -not always- it's shown 0%, after 5 or 10 seconds it takes the right read.

I'm gonna compile and flash and tell you

manusalgam, 2023-05-12:
Since I've flashed 1.4.0 I didn't noticed 0% read. Turned on/off about 20 or 30 times in different moments and it seems it's now OK

kb79000, 2023-05-12:
> Since I've flashed 1.4.0 I didn't noticed 0% read. Turned on/off about 20 or 30 times in different moments and it seems it's now OK

Hello how do you get version 1.4.0 ? Thank

manusalgam, 2023-05-12:
> > Since I've flashed 1.4.0 I didn't noticed 0% read. Turned on/off about 20 or 30 times in different moments and it seems it's now OK
> 
> Hello how do you get version 1.4.0 ? Thank

I've uploaded here: https://endless-sphere.com/sphere/threads/bbs-fw-open-source-firmware-for-bbshd-bbs02-controller.117092/page-10

kb79000, 2023-05-12:
> I've uploaded here: https://endless-sphere.com/sphere/threads/bbs-fw-open-source-firmware-for-bbshd-bbs02-controller.117092/page-10

Thx Is it possible to have the technique to compile the source ?😉

manusalgam, 2023-05-12:
> > > > Since I've flashed 1.4.0 I didn't noticed 0% read. Turned on/off about 20 or 30 times in different moments and it seems it's now OK
> > > 
> > > 
> > > Hello how do you get version 1.4.0 ? Thank
> > 
> > 
> > I've uploaded here: https://endless-sphere.com/sphere/threads/bbs-fw-open-source-firmware-for-bbshd-bbs02-controller.117092/page-10
> 
> Thx Is it possible to have the technic to compile the source ?😉

There are instructions link in main page. You will need terminal software (make, used in terminal only) and visual studio for the config tool. It's not easy if you're not almost an advanced user

Therefore, I'll compile early versions and publish on endlesphere forum

kb79000, 2023-05-12:
> > > > > Depuis que j'ai flashé 1.4.0, je n'ai pas remarqué 0% de lecture. Allumé/éteint environ 20 ou 30 fois à différents moments et il semble que tout va bien maintenant
> > > > 
> > > > 
> > > > Bonjour, comment obtenez-vous la version 1.4.0 ? Remercier
> > > 
> > > 
> > > J'ai téléchargé ici : https://endless-sphere.com/sphere/threads/bbs-fw-open-source-firmware-for-bbshd-bbs02-controller.117092/page-10
> > 
> > 
> > Merci Est-il possible d'avoir la technique pour compiler le source ?😉
> 
> Il y a un lien d'instructions dans la page principale. Vous aurez besoin d'un logiciel de terminal (make, utilisé uniquement dans le terminal) et de Visual Studio pour l'outil de configuration. Ce n'est pas facile si vous n'êtes pas presque un utilisateur avancé
> 
> Par conséquent, je vais compiler les premières versions et les publier sur le forum endlesphere

Ok thx 😉

kb79000, 2023-05-12:
> Since I've flashed 1.4.0 I didn't noticed 0% read. Turned on/off about 20 or 30 times in different moments and it seems it's now OK
I just flash and I confirm that the battery bug at startup has disappeared, good work @danielnilsson9 😉

MrHumanRebel, 2023-05-30:
After compiling a beta FW including the implemented fixes, now my battery status works flawlessly, perfect solution.
Thank you!

danielnilsson9, 2023-05-30:
Sounds good, I'm closing this. 

I have also added a compile time option for mapping battery % to  displays having a non linear battery bar indicator, like SW102 and 800S (maybe?). I.e where 1 bar != <20%, 2 bar != <40% ...

If 800S is affected a mapping can be added if you report which percentages correspond to amount of bars on the display.

closed 2023-05-30 by danielnilsson9