danielnilsson9/bbs-fw #117: Config tool – x86 bit executable

Oxmaster, 2023-04-27:
Hello. I'd like to try the firmware on my BBS02B. 

My bicycle PC is oldish PC tablet with old Intel atom that only supports x86 OS. Can the config tool be compiled for x86 bit systems?

Also how to backup the stock firmware that is on the controller?

Thank you.

silenec, 2023-04-27:
It can't be backed up so I just made some screenshots, downloaded stock CRR1094824E010015.3_170815.hex (again, no way to test if it is compatible with your controller before you erase its memory) and hoped for the best. No idea about x86.

danielnilsson9, 2023-04-27:
Hi

Yes it is possible to compile for x86 too but its not part of the official release.
Backup of original firmware is not possible as @silenec said.

Please be aware of this bug if you are considering flashing on BBS02 as it can provide a bad experience:
https://github.com/danielnilsson9/bbs-fw/issues/114

Oxmaster, 2023-04-27:
In "flashing the firmware" section there is information:
`Proceed with caution. It is a good idea to have access to the original firmware if you for any reason want to roll back.`

Though that you can download your own firmware somehow based on that statement.

Well thanks anyway.

closed 2023-04-27 by Oxmaster

danielnilsson9, 2023-04-27:
Yes, official firmware can sometimes be retrieved from your bafang distributor or from forums on the internet where people have shared it. I would prefer not sharing it here to avoid any potential trouble.

People on endless-sphere can probably help you if your need it.

Oxmaster, 2023-04-27:
somehow managed to compile it for x86, but it's not as neat as single file and you need to install ".NET 6.0 Desktop Runtime". If somebody in the future will need the x86 version I've hosted it on my [google drive](https://drive.google.com/file/d/1AK0ehFl1Oa7IZhZheVWq46hS1rbMI8fr/view?usp=share_link) and added it as attachment. I have not tested it yet, will do tomorrow. 

[image]

[bbsFWtool-x86.zip](https://github.com/danielnilsson9/bbs-fw/files/11347513/bbsFWtool-x86.zip)