danielnilsson9/bbs-fw #123: Potential bug: Power (W) display only showing three digits

MrHumanRebel, 2023-05-08:
If occasionally the actual power is higher than 999 the display only shows 999 and not going past it. Was testing with PAS and with Throttle too, same behaviour. The original FW is I recall correctly could show up to 4 digits of (W).
BBS02B

Thanks in advance!

dav0000000, 2023-05-22:
That's a strange number for the firmware to stop at. I'd guess it's the display but you didn't mention the model number. The code is sending higher values for me.

MrHumanRebel, 2023-05-23:
My display is fine, I have a 800s display. 
With stock FW it shows 4 digit numbers, all the way with peaks to 1245Ws, now peaking all the time at 999W not going past it.

dav0000000, 2023-05-23:
You could put a hardcoded counter value in extcom to see if/when it stops at 999. A bit of a silly question, but watt whatage should be displayed, based on your sagged battery voltage and max current setting. Is it 1245 watts?

danielnilsson9, 2023-05-28:
No clue here, power is calculated by display. Only current is forwarded from this firmware which is in the resolution of 0.5A.
See:
https://github.com/danielnilsson9/bbs-fw/blob/23ab0ec635deaacc84aef9a447b30c46a5140d35/src/firmware/extcom.c#LL549C17-L549C18

I cannot see any error there. You have this display?
https://wooshbikes.co.uk/manuals/GSMkit-800S-lcd.pdf

Where is power shown? I cannot find any info about this

MrHumanRebel, 2023-05-29:
https://github.com/danielnilsson9/bbs-fw/assets/80947401/98af7e35-82ef-4b16-bb1d-bc05c7e3550e

MrHumanRebel, 2023-05-29:
Just to make sure my display is working fine, I flashed back a stock FW I found and then it shows up to 4 digits. 
As soon as I flashed back this cool FW it stops at 999 for some reason still. :(

danielnilsson9, 2023-05-29:
Thanks for confirming. This is a weird one, it happily goes past 1500W on my display. Must be something special with this display then. Maybe it expects a response to some message I haven't implemented, I don't know. 

Not much I can do but I will leave the issue open in case someone with access to this hardware would like to dig in and work on this.

- First thing to do would be to find out how it responds to different values in process_bafang_display_read_current.
- Next step would be to dump communication between display and controller with original firmware and see if there is something not yet known.

blind-oracle, 2023-08-05:
BTW on original FW other display (500C) behaves a bit differently with respect to power - instead of showing 999 it overflows and shows values like 50-100w when it's above some threshold.

I wonder how it will work with this custom firmware, time to flash :)

mgafton, 2023-10-20:
@blind-oracle I can confirm the same behavior also with my 500C display, it only shows 100W on max throttle, when it should be above 1000W,
It seems like the 500C display has multiple issues, I have posted the list of the encountered problems separately.