danielnilsson9/bbs-fw #126: Make PAS ignore pedal rotation during Walk Mode

majonessyltetoy, 2023-05-18:
When engaging Walk Mode my pedals rotate with the chainring (I know I should clean and lube the motor).
After releasing Walk Mode, since the pedals have been rotating PAS kicks in and send the bike (which I'm walking) into full throttle of whichever Assist Level it is in.

User solution: Go into Assist Level 0 before using Walk Mode or have a well maintained motor which does not rotate the pedals when the chainring rotate.

Developer solution: Make PAS ignore pedal rotation during Walk Mode and possibly have a timeout after Walk Mode like the original firmware.

majonessyltetoy, 2023-05-23:
Yes, but as soon as the user let go of the Walk Assist, then PAS reads the cadence pulse counter, which has been building up.

Would something like this work?
https://github.com/danielnilsson9/bbs-fw/commit/c96bd2e7586989ef265b70b96042907c4a9c8e5d

I don't have a build env to test it out.

dav0000000, 2023-05-23:
Not what you want to hear but the best solution for you it to buy and install a new pawl clutch (if that is the problem) and of course to setup a build environment. Your approach looks ok to me, you might want to stop the timer to update the value as it is a uint16_t.

void pas_zero_pulse_counter()
{
	ET0 = 0; // disable timer0 interrupts
	pas_pulse_counter = 0;
	ET0 = 1;
}

What if you just do this below in app.c without any other changes. Perhaps this would work too.

void app_set_assist_level(uint8_t level)
{
	if (assist_level != level)
	{
	 	if (assist_level == ASSIST_PUSH)
                 {
                        ramp_down_target_current = 0;     //not sure if this is actually necessary
                        system_delay_ms(1000);
                 }
		assist_level = level;
		eventlog_write_data(EVT_DATA_ASSIST_LEVEL, assist_level);
		reload_assist_params();
	}
}

aalm, 2023-05-26:
> I don't have a build env to test it out.

Have you managed to build it yet?
Was relatively easy to get the makefile working on linux(rpi4) for build env.

danielnilsson9, 2023-05-28:
I guess this could be a quite common issue as I have this problem on my 6 year old bbshd too. Will try to add a fix in next version.

danielnilsson9, 2023-05-28:
Implemented a 1 second timeout when releasing push walk, let me know how it works if anyone decides to test.

majonessyltetoy, 2023-06-03:
> Implemented a 1 second timeout when releasing push walk, let me know how it works if anyone decides to test.

Thank you. That fixed the issue :)

closed 2023-06-03 by majonessyltetoy