danielnilsson9/bbs-fw #127: THROTTLE_HARD_LOW_LIMIT_MV defined as octal == 320ul instead of 500ul

aalm, 2023-05-21:
Hi,

was fixing the formatting, and adding missing 'void's to prototypes etc. while i came across this:
https://github.com/danielnilsson9/bbs-fw/blob/7b40049afb47b965482b27f4159a8a28dc4debd0/src/firmware/throttle.c#L31

does seem like a possible bug to me?

dav0000000, 2023-05-22:
That would depend on the throttle. As you can change this value in the config tool it's probably not a bug even though it is lower than usual.

Does adding the voids change the compiled code in any way? I'm no c expert but perhaps the compiler doesn't care.

On a side matter. In motor.c there are some declarations that use int and could use bool. Would changing this save memory as memory is limited on the bbs02 and I think Daniel said there is only 4 bytes left.

dav0000000, 2023-05-22:
Please disregard my comment about the throttle. For some reason I thought that was the changeable parameter in cfgstore.c.

danielnilsson9, 2023-05-22:
Wow yes, I had not idea a leading zero ment octal in C. That is not intended, although it probably doesn't cause any real issue either.

aalm, 2023-05-22:
Yep. It just lowers the THROTTLE_HARD_LOW_LIMIT_ADC, so I was guessing your LOG_THROTTLE_ADC was possibly
inspired by this miscalculation at some point.

closed 2023-05-28 by danielnilsson9