danielnilsson9/bbs-fw #128: premature optimization or unnecessarily initialized variables?

aalm, 2023-05-26:
While I think it is usually good to initialize variables in C like has been done here, I don't quite like what I'm seeing in
the compiler output for assembly, but I'm also totally new to 8051-derivatives like stc15, so perhaps this doesn't matter?

So, the code is full of cases like this:

```
	uint8_t mode = OPERATION_MODE_DEFAULT;

	switch (msgbuf[2])
	{
	case 0x02:
		mode = OPERATION_MODE_DEFAULT;
		break;
	case 0x04:
		mode = OPERATION_MODE_SPORT;
		break;
	}

	app_set_operation_mode(mode);
```

where both cases of the switch do set the variable twice for (imo.) no good reason when compared to option below:

```
	uint8_t mode;

	switch (msgbuf[2])
	{
	default:
	case 0x02:
		mode = OPERATION_MODE_DEFAULT;
		break;
	case 0x04:
		mode = OPERATION_MODE_SPORT;
		break;
	}

	app_set_operation_mode(mode);
```

in this case the extra is practically this:
```
;	test.c:16: uint8_t mode = 0;
	mov	dptr,#_bafang_display_write_mode_mode_65537_13
	clr	a
	movx	@dptr,a
```

vs. no instructions, and the variable is just set within the switch.

Is this somehow possibly related to the "Memory Model" switch?
I mean, was this previously necessary to get the local variables into xram or something?

danielnilsson9, 2023-05-26:
No, just a matter of style I guess. There has been no effort to optimize this code anywhere except maybe some additional thoughts in the hot path of the TSDZ2 motor control interrupt.

On bbsx architecture the cpu is doing nothing for most of the time so it doesn't really matter

Only thing worth mentioning is that global direct variable initialization does not seem to work on large memory model, therefore all global variables are initialized in the "initialize" functions.

aalm, 2023-05-26:
> No, just a matter of style I guess. There has been no effort to optimize this code anywhere except maybe some additional thoughts in the hot path of the TSDZ2 motor control interrupt.
> 
> On bbsx architecture the cpu is doing nothing for most of the time so it doesn't really matter

Oh, I see. So I'm not going to bother with those, and this issue can be closed:) I like the style myself too over my example above.

> 
> Only thing worth mentioning is that global direct variable initialization does not seem to work on large memory model, therefore all global variables are initialized in the "initialize" functions.

Ok. Much appreciated info. I might look at the uart buffering soon, as I'm likely to get the bbs02.

closed 2023-05-26 by aalm