danielnilsson9/bbs-fw #135: Feature request: Timer option

MrHumanRebel, 2023-06-06:
I don't know if this is controlled by the FW or the display, so sorry if I my idea is stupid or pointless.
Currently as soon as you turn on the bike the timer clock starts and goes until you turn it off.
It would be nice to only count when the bike is actually moving, then giving the time spent riding, not the time spent the bike is turned on. The user would be able to choose which timer counter fits for the usecase. 
Is my idea feasable or make any sense?

danielnilsson9, 2023-06-06:
No sorry, that would be a function that is implemented in the display firmware. The controller firmware does not keep track of time or distance on this motor.

closed 2023-06-06 by MrHumanRebel

loligeman, 2023-09-18:
> I don't know if this is controlled by the FW or the display, so sorry if I my idea is stupid or pointless. Currently as soon as you turn on the bike the timer clock starts and goes until you turn it [...]

i know it's late BUT on my dpc 18 i had the same but with new EKD01/B02N bluetooth display it only tracks when riding, same with avg speed .